tomvandeneede/p2pp #86: PURGETOWERDELTA prints in mid air

gdombiak, 2020-12-30:
Hey guys,

Nice to see the project alive and kicking. Great progress has been made!!! It has been over a year since I played with multicolor printing so I'm slowly coming back. I'm using latest from [dev](https://github.com/tomvandeneede/p2pp/commit/083f7fd64edc22fa2a74d821b5d0a6d8e2cf3842) for my tests.

Here is the post-processed gcode: [E_PLA 201901MosaicKeychain1.mcf.gcode.zip](https://github.com/tomvandeneede/p2pp/files/5755345/E_PLA.201901MosaicKeychain1.mcf.gcode.zip). I'm using https://gcode.ws/ to analyze how things would print. Notice the following things:
1. Layer 32 is missing in the tower
2. Tower prints layer 42, then nothing and then layer 47

My start gcode in PS 2.2.0+
> ;P2PP PRINTERPROFILE=2efa7a3df34dabff
;P2PP SPLICEOFFSET=30
;P2PP MINSTARTSPLICE=100
;P2PP MINSPLICE=80
;P2PP MATERIAL_DEFAULT_0_0_0
;P2PP MATERIAL_PVA_PVA_0_0_0
;P2PP MATERIAL_PVA_PLA_0_0_0
;P2PP MATERIAL_PLA_PLA_0_0_0
;P2PP MATERIAL_PET_PET_0_0_0
;P2PP PURGETOWERDELTA=7
;P2PP CONSOLEWAIT
;P2PP LINEARPINGLENGTH=350
;P2PP EXTRAENDFILAMENT=200

Object being printed is the basic [keychain](https://www.thingiverse.com/thing:3301571)

Let me know if you need more info.
Thanks
Gaston

tomvandeneede, 2020-12-30:
Hi Gaston,

thanks for pointing this out.  Can you please provide the unprocessed gcode so I can try to repeat the problem and confirm the solution.

Thanks

gdombiak, 2020-12-30:
Hi Tom,

Here is the generated gcode before P2PP processes it: [E_PLA 201901MosaicKeychain1.gcode.zip](https://github.com/tomvandeneede/p2pp/files/5755443/E_PLA.201901MosaicKeychain1.gcode.zip)

Thanks,
Gaston

tomvandeneede, 2020-12-30:
looking into it.

tomvandeneede, 2020-12-30:
Social distancing between layers was applied... removed that to put layers closer together
thanks for pointing this out.... Fix in 6.2.5 and later

closed 2020-12-30 by tomvandeneede