tomvandeneede/p2pp #87: Error: float division by zero

PowerChaos, 2021-01-01:
P2PP code used;
`;Palette 2 Configuration
;P2PP PRINTERPROFILE=MY UID
;P2PP SPLICEOFFSET=20
;P2PP MINSTARTSPLICE=100
;P2PP MINSPLICE=80
;P2PP FULLPURGEREDUCTION
;P2PP AUTOADDPURGE
;P2PP LINEARPINGLENGTH=350
;P2PP EXTRAENDFILAMENT=550
;P2PP ABSOLUTEEXTRUDER
;P2PP MATERIAL_DEFAULT_0_0_0
;P2PP MATERIAL_PLA_PLA_0_0_0
;P2PP CHECKVERSION`

SPLICE OFFSET: 20.00mm
Full purge reduction configured
Linear Ping interval of 350.00mm
Extra filament at end of print 550.00mm
Convert to absolute extrusion parameters
Analyzing Layers / Functional blocks
Bed origin (10.0mm, 5.0mm)
Bed zise (190.0mm, 195.0mm)
`Analyzing tool loading scheme`
FULLPURGEREDUCTION activated
TOWERDELTA could not be applied to this print
Generate processed GCode

We're sorry but an unexpected error occurred while processing your file
Please sumbit an issue report on https://github.com/tomvandeneede/p2pp

Error: float division by zero

Traceback Info:
File "p2pp.py", line 88, in 
File "C:\Users\tomvandeneede\Documents\p2pp\p2pp-dev_Qt\test\_build_update_\p2pp\p2pp\mcf.py", line 779, in generate 
File "C:\Users\tomvandeneede\Documents\p2pp\p2pp-dev_Qt\test\_build_update_\p2pp\p2pp\mcf.py", line 358, in gcode_parselines 
File "C:\Users\tomvandeneede\Documents\p2pp\p2pp-dev_Qt\test\_build_update_\p2pp\p2pp\purgetower.py", line 134, in purge_create_layers 

this file is used
[https://www.thingiverse.com/thing:3478125](https://www.thingiverse.com/thing:3478125)

This is the normal output and why i use the above code
`SHORT SPLICE (min 80.0mm) Length:59.76 Layer:18 Input:3`
and goes on for the complete print ( all below 80 mm ) 

Thank you
Greetings from PowerChaos

tomvandeneede, 2021-01-01:
Hi,

could you please send me the gcode file?  Would like to check why the extrusion width is resulting to 0 apparently/

thanks!

PowerChaos, 2021-01-01:
Hi Tom,

here are 2 gcode files
one with out the extra code
the other with the extra code , so you can compare them

only setting that is changed is disabling the experimental no sparse layers and added the 2 lines of code
;P2PP FULLPURGEREDUCTION ;P2PP AUTOADDPURGE

for all other settings they are exact the same

Greetings from PowerChaos

[laptop.zip](https://github.com/tomvandeneede/p2pp/files/5758877/laptop.zip)

tomvandeneede, 2021-01-01:
Hi 

You have your extrusion Width set to 0.

Default is something like 0.45 or so for a .4 nozzle

Under printer settings”

6.2.6 has a check built in for this error but best is to make sure your profiles are correct.

Happy New year, happy printing

Greetings

Tom

PowerChaos, 2021-01-01:
Thank you
But in prusa slicer it is set at 100% ( 0.4 nozzle , 0.4 extruder ) 

maybe it takes 0 as value instead the auto value it is supposed to do ?
I also noticed before that a certain setting in % does not work ( in speed ) and that a fixed value need to be assigned

is it possible in the future to auto assign the values ( 0 is supposed to take above value ) in case those auto values get used ?

In a couple hours I can try it again as I am not home anymore ( new year )
Also a happy new year :)

greetings from PowerChaos

tomvandeneede, 2021-01-01:
It does now assign 0.45 (the default Prusa Value) and issue a log item

I suggest though to set the value for now as the default may vary from the assigned value
I could copy the extruder_size but isince people may nt be using 100%, this is arbitrarily as much as the .45

tomvandeneede, 2021-01-01:
Ok,

just recoded to take into account the %, what I think you need to do is add the % to the line/
At least in 2.3.0 RC it correctly shows the value but with a %

I added a check that in case of % it will calculate based on the nozzle width.   

I see that in case of 0 PrusaSlicer calculates the width, in that case I still will assume 0.45 which I think is ok.   This is only for the purge tower.

regards

PowerChaos, 2021-01-01:
This probably need a other issue but also related to those 2 settings
I changed the values to 0.4 so it generated with out a problem

print was going fine ( it created a skirt around the block ) and when it needed to change the color it created his first layer succesfull
Then the mess up part started
It suddenly started to extrude on full speed ( 150mm and more ) and dit not move
Kinda like a side purge
Then it moved and dit the same again
I needed to cancel the print as it was acting strange

I use a ender 3 V2 and palette 2s

As my prints contains short splices ( less then 80mm ) was this the only option I could use
But on that way it is kinda useless to use it 

can you check this please why it does that ? It’s the same file as above ( same settings exept I put extruder on 0.4 instead of 0/100% for all settings  ) 

thank you
Greetings from PowerChaos

tomvandeneede, 2021-01-01:
do you have the processed and unprocessed gcode so I can take a look

tomvandeneede, 2021-01-01:
Hi,

Did the test and ran your model in P2PP 6.02.06 (did some small other changes but unrelated

output spee speed of the tower is lower than the actual print speed. 

can you indicate where in the print this was happening?

PowerChaos, 2021-01-01:
the print happends around 2.30 hours ( or 16.2 meter fillament ) 
so right at the point it switched to the second color 
i going upload my mp4 file at the last 2 seconds you can see the error

before i forget, this is the setup i use

Windows 10 + P2PP exe file ( from homepage )
`P2PP Version: 6.02.05
Python Version: 3.9.1
Version Check: Current version 6.02.05 (Dev version up to date) ` 
Prusa Slicer 2.3.0-RC2
Octoprint ( python 3.7 )  converted to Canvas Hub
and a few minor plugins that has no effect on the print itself ( bedlevel, The spagetti detective,  tool to calculate exact print time)

[Laptop_Pieces3_16h12m_20210101-134513.mcf.gcode.zip](https://github.com/tomvandeneede/p2pp/files/5759302/Laptop_Pieces3_16h12m_20210101-134513.mcf.gcode.zip)

because the file is 80M big (raw file ) is here the link
https://powerchaos.com/p2pp/
in case you do not trust the link, i used the domain that contains my own name :D

Greetings from PowerChaos

tomvandeneede, 2021-01-01:
6.02.07 should solve this issue.   there was abug in the processing of Absolute extrusion conversion during the conversion of full purge

Regards

PowerChaos, 2021-01-01:
ok thank you, it is working now

there is 1 more thing that is kinda needed

i am now test running with the multi key ( calibration tool - quick print ) 
and i noticed the first switch it does fine
but with the second switch , when there was a layer difference it move from 0.4 to 2.4 after it is near the print
with result , it crashes into my print first before it does the Z height ( or it does it at the same time , numbers are wild guesses ) 

that was probably also the thing it dit with my other print ( when it was closer to each other )
can you change that behavior so it first go up back to where it was ( maybe slower ? ) before it start to move on the XY axes

using latest build now, everything else works fine

please see video of what i mean

https://user-images.githubusercontent.com/7650333/103445637-30e40080-4c77-11eb-8f41-0bbc39c3c14b.mp4

Greetings from PowerChaos

tomvandeneede, 2021-01-01:
Will look at this in the morning.

Sorry you have all these issues but I’m sure all gets sorted quickly

Regards
Tom

Verstuurd vanaf mijn iPad

tomvandeneede, 2021-01-01:
Could you please supply the gcode (processed and unprocessed from the print)
There are tons of different settings and it is impossible to solve issues without proper possibility to repeat them

Thans a lot

T

Verstuurd vanaf mijn iPad

tomvandeneede, 2021-01-01:
Had a look and made 6.2.8 that possibly solves the issue, but would be nice to verify on your gcode as matter of test

PowerChaos, 2021-01-01:
@tomvandeneede no problem :)

I going attach my printer profile tomorow ( full Config ) so it is easyer to reproduce

I work with octoprint so I do not have unprocessed files. I use a stl file ( multicolor in this case ) and get directly uploaded to octoprint 

I followed the pdf to Config the code to get p2pp to work
But as it trows errors at splices that are to small so I am forced to use the “special” mode so my splices are atleast 80mm

I tried it to fix it with multiple instances of same print ( full bed -> less waste tower and normaly longer splices  ) but that dit not solve it

I switched to canvas but that slicer ruins my prints badly ( lot of random Seams and a raw surface on all my prints. Just a big mess ) so I am happy this project exist so I can keep using my favorite slicer :)

I am happy to help to make this program a great program
So far with the “non experimental” settings does it work great. Sadly that I need that 1 function to get my splices correct ( unless it could be forced for the other modes ? So it always have the min splice amount even with normal waste towers ) 

anyway 
Tomorow I will test your new version ( just saw when I was typing that you released a update ) and let you know if it works 
Also going provide all the needed files to make testing a lot easyer on your end 

greetings from PowerChaos

tomvandeneede, 2021-01-01:
you can add a line to the config to store the unprocessed files, but then you need to save them not send them to octoprint directly

add 
;P2PP SAVEUNPROCESSED

to your startup gcode

PowerChaos, 2021-01-02:
Thank you Tom,
it works now ( for the test print , now printing some other longer test ) 

only thing i noticed is the skirt , it seems to mess up that skirt ( and change color for the skirt , probably just he file related ) 
but atleast it works fine now 👍 

for the test print i made , it seems the settings also make it a little overlap ( the wiping block wiht the print ) 
if i move it a litle bit then it says that it is a side wipe 

but those are just minor things and probably related to my settings and the placement of the things

anyway
here are my settings file and here is also the file to test
[logo.zip](https://github.com/tomvandeneede/p2pp/files/5760177/logo.zip)
[PowerChaos.zip](https://github.com/tomvandeneede/p2pp/files/5760178/PowerChaos.zip)

Logo.zip contains my logo converted from a image and splitted in multi objects with prusa slicer ( and added a background with tinkercad ) 
it also contains a MF file , that is the latest project file i exported

PowerChaos.zip contians the settings of the slicer , nohting has beein changed so it should be exactly like i have it now

Thank you for your fast response and fast solution
Greetings from PowerChaos

sidenote:
the Saveunprocessed dit not seem to work , i exported 2 files , one with that command included and one with out but size is exactly the same

tomvandeneede, 2021-01-02:
Hi,

the overlap is something I can look into for sure.   I am rethinking the placement of the power tower with another project.  Will integrate that in p2pp hopefully soon.
what do you mean mess up the skirt?

PowerChaos, 2021-01-02:
on both test prints
it create first the square for the wipe tower
then it create a skirt directly true the whipe tower, it does not allign itself anymore
then on the second loop it goes again out of sync
and with the multicolor key it even goes complete out of sync ( creating 2 color skirt instead normal one )

it is only the skirt , as the print is just fine 
its hard to see on a timelaps , but maybe you can see it on there

https://user-images.githubusercontent.com/7650333/103462148-a4434c00-4d23-11eb-8a12-eec29d04b3a0.mp4

tomvandeneede, 2021-01-02:
which ode is this? Full purge?

> On 2 Jan 2021, at 17:56, PowerChaos <notifications@github.com> wrote:
> 
> https://user-images.githubusercontent.com/7650333/103462148-a4434c00-4d23-11eb-8a12-eec29d04b3a0.mp4 <https://user-images.githubusercontent.com/7650333/103462148-a4434c00-4d23-11eb-8a12-eec29d04b3a0.mp4>

PowerChaos, 2021-01-02:
same as before
`;P2PP FULLPURGEREDUCTION
;P2PP AUTOADDPURGE`

sorry for late reply ( i was eating ) 
i need to use the auto add purge function as all my prints seems to have less then 80mm of fillament for some reason

Greetings from PowerChaos

tomvandeneede, 2021-01-02:
check your purge volumes  they may be defaulted to the 70 values which is not enough

PowerChaos, 2021-01-02:
i use the default of 70/70
i got no idea to what else i need to set it :( , i like to keep the waste pile as small as possible ( specialy with vertical multi color letters, last print took 14 hours and 300% was my waste tower -> canvas ^^  ) 

or is this related to the extra amount of fillament that it use to extrude? and not related to how mutch color swap it use ?

tomvandeneede, 2021-01-02:
oh you need test the 70 to 180 or something to get rid of the short splices and even higher if you have color bleeding
this number defined the amount of purge between two colors
differs from filament to filament and color to color
read to any other needs a lot more than e.g. white to another color

PowerChaos, 2021-01-02:
so far with the auto purge option it works fine and waste amount is just perfect 🥇 
i print in the following collors

black -> silver -> red -> gold
and with the latest test print only the 3rd last layer got a big mess up ( red became black , black became silver and silver became red ) until next round it got fixed again
looked like a wrong offset for that single layer ^^ ( pings are also 107% and pongs 98% ) 

[image]
[image]
[image]

tomvandeneede, 2021-01-02:
hold on to those colors…. they really swap out easily
with red white it can take 3 to 4 times as long

PowerChaos, 2021-01-02:
because you asked it
this is from black to Glow in the dark ( transparant green glow ) 

[image]
seems also fine for me 
was with 2 versions ago ( after the bump error )

or i just need to say that the auto function works great :D `;P2PP AUTOADDPURGE` <-- Nice ^^

tomvandeneede, 2021-01-02:
thanks!   looking nice… apparently not glow in not-dark as well :-)

BTW where are you from?

PowerChaos, 2021-01-02:
i am from belguim ( https://vpsdata.be is my shop/company ) 

and i fresh started with 3D printing ( i mainly repair pc's and got a game center , 3D printing is a new thing i added because of covid ) 

thats why i also love your work , as on first try of the multi color it was a big mess until i found this plugin
now the printing are nice and so far it works great, just sorting out the few bugs and then i can do my long prints ( 2 days printing ) 
and on positive side , i do not need to mess around with other slicers exept with prusa for fine tweaking ( is already nicely tweaked for me ^^ )

tomvandeneede, 2021-01-02:
ha dan gaan we dat hier in’t nederlands oplossen niet :-)

PowerChaos, 2021-01-02:
dat mag van mij ook, anders moet je op mijn discord komen kan je daar direct chatten ^^
dacht van dit programma is internationaal dus bugs worden in het engels gezet

ik ben momenteel begonnen met de 2de test bestand ( met meer offset ) dus zal zien wat dit bestand zal geven

mvg
PowerChaos

tomvandeneede, 2021-01-02:
Wat is je discord account?

PowerChaos, 2021-01-02:
[VPS Data] PowerChaos
#5274
ofwel via deze link
https://discord.gg/ZwJ8epu

mvg
PowerChaos

closed 2022-02-05 by tomvandeneede