tomvandeneede/p2pp #88: Bed size wrong

PowerChaos, 2021-01-02:
Hi,
this is a other bug report

the bed size setting get ignored

this is my setting
```
;P2PP BEDORIGINX=0
;P2PP BEDORIGINY=-10
;P2PP BEDSIZEX=235
;P2PP BEDSIZEY=235
```
[image]

and this is the response ( with errors ofcourse )
```
P2PP Version: 6.02.08
Python Version: 3.9.1
Reading File C:\Users\Staff\AppData\Local\Temp\.PrusaSlicer.upload.ca19-ccba-61e7-49d1
Analyzing Prusa Slicer Configuration
All purge lenghts 70/70 OR 140. Purge lenghts may not have been set correctly.
SPLICE OFFSET: 30.00mm
Full purge reduction configured
Linear Ping interval of 350.00mm
Extra filament at end of print 550.00mm
Convert to absolute extrusion parameters
Analyzing Layers / Functional blocks
Manual bed size override, PrusaSlicer Bedshape configuration ignored.
Bed origin (15.0mm, 10.0mm)
Bed zise (190.0mm, 195.0mm)
`Analyzing tool loading scheme`
SIDEWIPE and SKIRTS are NOT compatible in PS2.2 or later
Side wipe activated
FULLURGEREDUCTION is incompatible with SIDEWIPE, parameter ignored
Generate processed GCode
Using sidewipe with undefined SIDEWIPELOC!!!
Generating GCODE file: C:\Users\Staff\AppData\Local\Temp\.PrusaSlicer.upload.ca19-ccba-61e7-49d1
```

tomvandeneede, 2021-01-02:
yep it is taken from ps bed definition now  can you pls share your gcode

tomvandeneede, 2021-01-02:
really need gcode files with all of these issues :-)   there are so many settings that trying to figure out what is causing the exact problem is difficult with the supplied info only

PowerChaos, 2021-01-02:
[Logo-2_28m_20210102-175825.mcf.gcode.zip](https://github.com/tomvandeneede/p2pp/files/5760564/Logo-2_28m_20210102-175825.mcf.gcode.zip)

that is why i gave you my full profile :P  but here is the code

Greetings from PowerChaos

tomvandeneede, 2021-01-02:
Hi 

Think I got it sorted

the full profile has issues that I clobber up my own config :-(   I have had numerous failed prints due to accidentally referring to a wrong profile
hence I prefer gcode reading

Bed processing should be automatic and I added an extra check and seems to be fine

Concerning the skirt, I have added a warning when skirt and fullpruge are used together for now.
hope to find a workaround soon, but forbittinf the skirt is likely the best option
You can always auto add purge to make up for short splices

PowerChaos, 2021-01-02:
the skirt get used in my case for the first line, as it start extracting only after a few mm
normaly i use the build in "first line" but was not sure if it works with this plugin ( founded in docs that it should work ) 

thank you for your quick fixing 😄 
and for the gcode , next time i will add the gcode to my reply
for the bug i splitted this one up as it is not really related to the other bug

if i find more "bugs" i let you know

Thank you
Greetings from PowerChaos

tomvandeneede, 2021-01-02:
thank you for reporting all this
really helps a lot ironing out issue

PowerChaos, 2021-01-02:
update
the bed size error is gone
but now it just plain ignore the bed size settings
see Gcode and image :P

[image]
[image]
[debug-Logo-2_28m_20210102-193100.mcf.gcode.zip](https://github.com/tomvandeneede/p2pp/files/5760703/debug-Logo-2_28m_20210102-193100.mcf.gcode.zip)

tomvandeneede, 2021-01-02:
hmm let me see

> On 2 Jan 2021, at 19:33, PowerChaos <notifications@github.com> wrote:
> 
> the bed size error is gone
> but now it just plain ignore the bed size settings
> see Gcode and image :P

tomvandeneede, 2021-01-02:
intention is to obsolete bedside and take it from the printer definition.
what bed size does it report?

PowerChaos, 2021-01-02:
this zip file will be my next test , changed the bed size offset and the report is a bit better ( unless it goes back to 0*0 then its just smaller xD ) 

it contians a few images to make things easyer
[p2pp.zip](https://github.com/tomvandeneede/p2pp/files/5760727/p2pp.zip)

i dit not test this print yet , but for the report answer
it report 220 * 220 and it extract 15 * 20
even when my settings was 0*-10
and in prusa my settings are 220*220 with -15 offset
but it just goes to the 0*0 value of my bed (even past the prime line ) 

i let my test run as i also use it for my color calibration 

Greets from PowerChaos

tomvandeneede, 2021-01-02:
this is correct processing

5mm is added on the y-axis as this ie needed to cope with the introduction line printed by prusa.
there is no easy way to take this our on other printers.  Technically it should be harmless

PowerChaos, 2021-01-02:
de nieuwe code heb ik geprobeerd
en zoals ik al had gedacht , de offset code werkt niet
die blijft op 0x0 staan ongeacht welke instelling je gebruikt
dit is de print van bovenste file
dit is het resultaat
[image]

closed 2022-02-05 by tomvandeneede