tomvandeneede/p2pp #89: how do I update my mac version of p2pp to latest?

audiorazor, 2021-01-03:
I see that the p2pp.zip is removed for the mac.   I managed to get your old way of installing to work by updating python.  Any chance I can still just download that file?

tomvandeneede, 2021-01-03:
new version is distributes as a dmg file with all included, just unpack to your Applications folder and let it run
the zip file will no longer work due to some recent changes in macOS Big Sur
it is slightly unfortunate but adding another delivery method is currently not really sustainable for me.

audiorazor, 2021-01-03:
is this from the dev branch?  im not seeing it

tomvandeneede, 2021-01-03:
The link to the distruibutions is on the dev readme page at the top in big letters
you should be able to download the latest dev version there in the new format

closed 2021-04-03 by audiorazor