tomvandeneede/p2pp #90: Code doesn't match Dropbox releases?

aschmitz, 2021-01-22:
It looks like there are a bunch of changes on the Dropbox folder that aren't tracked in this Git repository, including two major versions and likely a bunch of point releases. Is the code for those releases available?

(I know the Windows .zips are largely bundles of .pycs, and could be reverse-engineered, but I'd prefer to avoid doing that if necessary. I'm also on Linux, and it seems easier to pull the repository and run from there than to extract all of the necessary files from the Windows release.)

Thanks!

tomvandeneede, 2021-01-22:
All changes in the zip/app files are also in the repository.   You can perfectly still use the repository as before but you will need to install PyQT5 and the PIL libraries
I unfortunately do not have the bandwidth to support an extra set of semi binary releases, specially since there are a few linux flavours making things not easier.
but technically if you install prerequisites, it should run from the dev-release just as well

aschmitz, 2021-01-22:
Ah, this was just user error: I was on the `master` branch as the GitHub default, and not the `dev` branch. Sorry about that! (Might be worthwhile to switch the GitHub default branch, given, but it's my fault for not investigating further.)

Thanks!

closed 2021-01-22 by aschmitz

tomvandeneede, 2021-01-22:
no problem..