tomvandeneede/p2pp #91: 17 character PRINTERPROFILE

maziggy, 2021-01-26:
My PRINTERPROFILE has 17 characters instead of 16. Therefore Palette says that it has detected a new printer. 

Am I doing something wrong or did it change?

Thx.

tomvandeneede, 2021-01-26:
can you please send me the gcode file

maziggy, 2021-01-26:
I guess you mean the start GCODE.

`;Palette 2 Configuration 
;P2PP PRINTERPROFILE=Dd6bb61e8c8c75e63
;P2PP SPLICEOFFSET=40
;P2PP MINSTARTSPLICE=120
;P2PP MINSPLICE=100
;P2PP LINEARPINGLENGTH=350
; tool purge length information for p2pp
;P2PP TOOL_0=PLA:[120;120]
;P2PP TOOL_1=PLA:[120;120]
;P2PP TOOL_2=PLA:[120;120]
;P2PP TOOL_3=PLA:[120;120]
; Material splicing profile information
;P2PP MATERIAL_DEFAULT_1_1_1
;P2PP MATERIAL_PLA_PLA_1_1_1
;P2PP EXTRAENDFILAMENT=200

G21                                         ; metric values
M290 R0 S0		   ; reset baby stepping
G90                                         ; set absolute positioning
G28                                         ; home all
M561			; Disable any current bed compensation
G32
M564 S0		; disable axis limits
G1 X150 Y150 F6000
G90 G1 Z3 F400
M400
G92 Z4 			; Reset z to 8 to allow jogging up to touch bed to nozzle
M291 P"Jog the Z Axis until the bed and nozzle are touching and click OK" R"Setting Z=0" Z1 S3
M400
G92 Z0			; Set z = 0
G90 G1 Z3 F400
M400
G1 Z5
M400
G30 S-1
M400
G1 Z5
M400
G30 S-1
M400
G1 Z5
M400
G30 S-1
M400
G1 Z5
M400
G30 S-1
M400
G1 Z5
M400
G30 S-3 ; measure probe height and set it.
M400
M500 P31 ; save trigger height to config-override.g
G1 Z5
M400
M564 S1		; enable axis limits
M280 P0 S160 ; reset Bltouch in case of errors
M557 X29:316 Y14:310 S20:20  ; mesh grid dummy
G29		   ; mesh bed leveling
G29 S1		   ; load bed mesh
M221 S100		   ; set extrusion factor
G92 E0 ; reset extrusion distance
G1 X12 Y20 Z0.3 E5 F6000
G1 X12 Y30 F6000
G1 Y210 E20 F1000
G92 E0 ; reset extrusion distance
`

tomvandeneede, 2021-01-26:
Don’t copy the capital D

> On 26 Jan 2021, at 15:35, maziggy <notifications@github.com> wrote:
> 
> d6bb61e8c8c75e63

maziggy, 2021-01-26:
Ahhhh OK, will try. Thanks for the quick reply!

closed 2022-02-05 by tomvandeneede