tomvandeneede/p2pp #93: Feature Request: Support Tool Changers and Multiple Palettes

benwithem, 2021-01-27:
# Description
Proposing support for tool changer printers where each tool could potentially have a Palette with 4 virtual extruders assigned to each tool.

# Extruder Assignment
A default assignment of extruders to tools could the first 4 extruders would be assigned to T0, and the next 4 to T1, etc. Or a simple parameter.

```
;P2PP TOOLCHANGER=4
;P2PP TOOLCHANGER1= [1,2,3,4]
;P2PP TOOLCHANGER2= [5,6,7,8]
;P2PP TOOLCHANGER3= [9,10,11,12]
;P2PP TOOLCHANGER4= [13,14,15,16]
```

# Purge Strategies
A default wipe tower would allow for color changes on virtual extruders assigned to the same physical tool head. This would not be appropriate however for purging color changes that span different physical tools.

# Parsing GCode For Different Palettes
This seems to be the biggest risk. How to control 4 different Palettes in a connected mode. Sadly I haven't been able to dive into this yet.

tomvandeneede, 2021-01-28:
Sounds interesting, BUT…

1/ how would you set this up in Prusa?   The tower is entirely defined in PrusaSlicer, I am not aware that there is a way to not do a tower between color changes
2/ would each tool changer have a P2 attached to it? depending on the technology used there would be extension of all functionality (all calculations) over the different infivitual hardware tools
3/ even if this would work, you need to implement this in the plugin as it needs to handle multiple palettes on the same hub

if this is for none-palette changers, it might work likely other control sequences need to be inserted that would not work on P2

let me know how you would address these challenges

benwithem, 2021-01-28:
I updated the FR description a bit to cover a few of these things but here are the answers as I currently understand them.

1) I believe we would need to not generate or ignore the purge tower and use a different mechanism like a side purge, or bucket. Probably a similar solution to the existing code for the BigBrain.
2) Each tool would have a P2 attached to it, and yes this makes all the calculations more difficult.
3) I have opened a FR with Mosaic here attempting to address some of these things, here is the summary:

https://gitlab.com/mosaic-mfg/palette-2-plugin/-/issues/16

On the P2PP side we would need to send something to the plugin to let it know which P2 to queue commands to, something like this:

```
OXX UNIQUE_P2_IDENTIFIER ; Start sending commands to this P2
...
...
...
OXX UNIQUE_P2_IDENTIFIER ; Stop sending commands to this P2
```

At an extremely high level the following changes would be needed in the Octoprint Plugin.

- Multiple P2 connections and a way to uniquely identify them
- Start sequence for multiple P2s
- Logic to decide which P2 to queue a command to
- Changes to existing counters, ping/pong, heartbeat, splice correction to account for multiple P2s

tomvandeneede, 2021-01-28:
1/ ok understand and is fairly easy
2/ not very difficult BUT you will need to find a way to have multiple configuration sections of P2 in the output file so also the plugin on the Octopi will need to change
3/ sounds reasonable still there is a  state type machine maintained in the plugin that will need to be multiplied… 

some other topics that need addressing

* loading of the filament, specially when mixing non-P2 with P2 loaded interfaces
* given that you will need to purge when switching colors within the same tool, you will lose a lot of the optimisation  as the slicer it not meant for that 
* how will you setup multiple extruders with different X/Y offsets or will you use a very precise tool changer that has every physical tool in the same place if you have an X/Y offset with the tools I don’t think you can programme that in the PrusaSlicer unless you manually start calculating offsets etc… interesting yet a lot of things can go wrong

guess the overall big question for me is…. how much development work will go into this (continually…. as this continues extra testing effort after changes) versus the number of actual users\
don’t get me wrong I have thought about adding this more than once but it gets quite complex with the already added more than 4 color print functions that are still under development

Let’s wait and see how Mosaic picks up on your suggestion

benwithem, 2021-01-28:
I think we are on the same page as far as the level of effort and the where the majority of the difficulty lies.

For reference, this is the tool-changer I had in mind for this: https://e3d-online.com/products/toolchanger-motion-system-bundle-tools Imagine 4 E3D Hemera Direct Drive Extruders all with P2s attached, could even just setup 2 P2s and only print with 2 tools.

I agree with waiting to see what kind of response comes from Mosaic, and very thankful for your engagement in this thought experiment!