tomvandeneede/p2pp #94: TypeError

maziggy, 2021-01-29:
I'm using the Master branch under MacOS 11.1 and get the following error:

`Post-processing script /Users/martin.ziegler/tools/p2pp/p2pp_dist/p2pp.command on file /Users/martin.ziegler/Downloads/p2pp/foo.prusa.mcf.gcode failed.
Error code: 1
Output:
Traceback (most recent call last):
File "/Users/martin.ziegler/tools/p2pp/p2pp_dist/P2PP.py", line 146, in <module>
main(vars(arguments.parse_args()))
File "/Users/martin.ziegler/tools/p2pp/p2pp_dist/P2PP.py", line 87, in main
mcf.generate(v.filename,
File "/Users/martin.ziegler/tools/p2pp/p2pp_dist/p2pp/mcf.py", line 961, in generate
gcode_parseline(process_line_count)
File "/Users/martin.ziegler/tools/p2pp/p2pp_dist/p2pp/mcf.py", line 570, in gcode_parseline
if not coordinate_on_bed(_x, _y):
File "/Users/martin.ziegler/tools/p2pp/p2pp_dist/p2pp/mcf.py", line 178, in coordinate_on_bed
return x_on_bed(x) and y_on_bed(y)
File "/Users/martin.ziegler/tools/p2pp/p2pp_dist/p2pp/mcf.py", line 174, in x_on_bed
return inrange(x, v.bed_origin_x, v.bed_origin_x + v.bed_size_x)
File "/Users/martin.ziegler/tools/p2pp/p2pp_dist/p2pp/mcf.py", line 166, in inrange
return low <= number <= high
TypeError: '<=' not supported between instances of 'float' and 'str'`

GCODE file is attached.

Thanks!
[gcode.zip](https://github.com/tomvandeneede/p2pp/files/5893827/gcode.zip)

maziggy, 2021-01-29:
In the meantime I've found out that the start GCODE line:

M557 X29:316 Y14:310 S20:20  ; mesh grid dummy

is parsed and the string in question is '29:316'.

tomvandeneede, 2021-01-29:
What pronter is this for?

maziggy, 2021-01-29:
A BLV mgn Cube with 330x330 bed.

tomvandeneede, 2021-01-29:
to my knowledge the dev version should will solve this

maziggy, 2021-01-29:
It does ;) Thanks!

closed 2022-02-05 by tomvandeneede