tomvandeneede/p2pp #97: Attempting to set up P2PP with a Palette +

malcinator, 2021-03-15:
I'm getting this error when I try and export using PrusaSlicer.

Error: can only concatenate str (not "bytes") to str
Traceback Info:
File "/Applications/P2PP.app/Contents/Resources/P2PP.py", line 88, in 
File "p2pp/mcf.pyc", line 830, in generate 
File "p2pp/gui.pyc", line 142, in log_warning 

I have a Palette+ and understand that this is still in the experimental stage. I have downloaded the very latest ver of P2PP from dropbox and have (as best as I can tell) configured profiles in PrusaSlicer.

tomvandeneede, 2021-03-15:
could you please send me the gcode file
that will simplify the debugging

malcinator, 2021-03-15:
Please let me know if I have my profiles all screwed up!

[Mosaic_Test_2_PP.mcf.gcode.zip](https://github.com/tomvandeneede/p2pp/files/6141036/Mosaic_Test_2_PP.mcf.gcode.zip)

tomvandeneede, 2021-03-15:
which version of P2PP are you using.

malcinator, 2021-03-15:
P2PP Version: 6.02.24
Python Version: 3.9.1

tomvandeneede, 2021-03-15:
I see, the file you sent arrived empty, would you mind sending with Facebook messenger

closed 2022-02-05 by tomvandeneede