tomvandeneede/p2pp #100: Not extruding immediately after side purge

jessegrunert, 2021-03-25:
Hello Tom,

I found another issue this morning, here are the details.

When using the side purge functionality, immediately after the side purge is finished, extrusion is not happening for a brief period when going back to printing. I have tried this both with/without AUTOADDPURGE, the result is the same.

These are the relevant settings:

```
;P2PP WIPEFEEDRATE=200
;P2PP SIDEWIPELOC=X-15
;P2PP SIDEWIPEMINY=120
;P2PP SIDEWIPEMAXY=120
;P2PP AUTOADDPURGE
```

Here is what I'm getting in the g-code and where I think the problem is. The E-3.00000 commands are causing the printer to go back to printing while no filament is at the nozzle.

```
; [--P2PP-- side wipe skipped (NO_E)] - G1 E-0.40000 F3600
; [--P2PP-- side wipe skipped (NO_E)] - G1 Z0.800 F7800
;---------------------------
;  P2PP SIDE WIPE: 118.962mm
G1 F8640
G1 X-15.000 Y120.000
G1 E50.00000 F200
G1 E-3.00000
G1 E0.38000 F200
G1 E50.00000 F200
G1 E-3.00000
G1 E0.38000 F200
G1 E18.96213 F200				<--- positive filament at nozzle
G1 E-3.00000 					<--- filament -3mm from nozzle
G1 E0.38000 F200				<--- filament -2.62mm from nozzle
G1 E-0.38000					<--- filament -3mm from nozzle again
G1 F8640
;---------------------------
G1 X114.793 Y137.887
G1 Z0.200
G1 E0.38000 F3600				<--- filament -2.62mm from nozzle
;TYPE:Perimeter
;HEIGHT:0.2
G1 F2400
G1 X114.682 Y137.915 E0.00325 	<--- Regular extrusion starts without filament at nozzle
G1 X114.634 Y137.906 E0.00137
G1 X114.548 Y137.841 E0.00305
```

I have attached the whole file for reference as well.

-Jesse
[201901MosaicKeychain1.mcf.gcode.zip](https://github.com/tomvandeneede/p2pp/files/6207483/201901MosaicKeychain1.mcf.gcode.zip)

tomvandeneede, 2021-03-25:
Hi Jesse,

 

Can you do me a favour and send me the unprocessed gcode for this one and I will have a look.

 

 

regards

jessegrunert, 2021-03-25:
Please see attached. Also a screenshot of my layout so you can see where I have the purge tower. To get you the unprocessed file I just disabled the output option to send to p2pp.exe and saved the gcode. Is that what you're looking for?

[201901MosaicKeychain1.mcf.gcode - unprocessed.zip](https://github.com/tomvandeneede/p2pp/files/6207619/201901MosaicKeychain1.mcf.gcode.-.unprocessed.zip)

[image]

jessegrunert, 2021-03-26:
Please see above PR for a potential fix!

tomvandeneede, 2021-03-26:
Jesse,

looked at your solution, and it looks solid but would need a lot of testing in all other scenarios.

I opted to just hardcode the large retract/unretract in the code directly:
issue_code("G1 E{:.5f} F{}".format(sweep, v.wipe_feedrate))
issue_code("G1 E-3.0000 F300")
issue_code("G1 E3.0000 F200")

this should give the same result without any unpleasant side effects in other places

I will upgrade this in a next release due somewhere next week.

jessegrunert, 2021-03-26:
Hi Tom,

A fair assessment there, your solution is much simpler :) I'll look forward to your next release :)

-Jesse

closed 2021-03-26 by jessegrunert