tomvandeneede/p2pp #99: string index out of range

jessegrunert, 2021-03-25:
Hello!

I am just getting back into using P2PP again and have upgraded to the latest version 6.02.29. I received this error when trying to output g-code via PrusaSlicer 2.3.0-beta2

`
P2PP Version: 6.02.29
Python Version: 3.9.1
Reading File C:\Users\Jesse\Downloads\Gustav_the_Turtle_multi-material\files\test.gcode
Analyzing Prusa Slicer Configuration
SPLICE OFFSET: 20.00mm
Extra filament at end of print 100.00mm
Extrusionwidth set to 0, defaulted back to 0.45
Analyzing Layers / Functional blocks
Manual bed size override, PrusaSlicer Bedshape configuration ignored.
Bed origin (0.0mm, 0.0mm)
Bed zise (240.0mm, 240.0mm)
Side wipe activated
Generate processed GCode

We're sorry but an unexpected error occurred while processing your file
Please sumbit an issue report on https://github.com/tomvandeneede/p2pp

Error: string index out of range

Traceback Info:
File "p2pp.py", line 88, in 
File "C:\Users\tomvandeneede\Documents\p2pp\p2pp\_build_update_\p2pp\p2pp\mcf.py", line 804, in generate 
File "C:\Users\tomvandeneede\Documents\p2pp\p2pp\_build_update_\p2pp\p2pp\omega.py", line 104, in header_generate_omega 
File "C:\Users\tomvandeneede\Documents\p2pp\p2pp\_build_update_\p2pp\p2pp\omega.py", line 179, in header_generate_omega_palette2 
File "C:\Users\tomvandeneede\Documents\p2pp\p2pp\_build_update_\p2pp\p2pp\colornames.py", line 77, in find_nearest_colour 
File "C:\Users\tomvandeneede\Documents\p2pp\p2pp\_build_update_\p2pp\p2pp\colornames.py", line 64, in color2rgb 
`

tomvandeneede, 2021-03-25:
can you  please send me the gcode file you were running

jessegrunert, 2021-03-25:
Here you go, I had to zip it for upload.

[test.zip](https://github.com/tomvandeneede/p2pp/files/6202587/test.zip)

tomvandeneede, 2021-03-25:
; extruder_colour = #000000;;;

looks like your extruder color is not set.   Not sure why this is as normally there is always a value there
actually the first time I see this

jessegrunert, 2021-03-25:
Hello, thanks for letting me know that, I was able to find the problem with that knowledge. As an FYI, if someone else ever sees this, here was the problem. When I added additional extruders, the color is initially undefined, I needed to add a color for each extruder, then the error went away.

[image]

closed 2021-03-25 by jessegrunert

tomvandeneede, 2021-03-25:
Jesse,

thanks for sharing this information with me.   I think this only was introduced in the latest versions
Anyway in a next release I will add explicit error checking for this so it does not occur

thanks again for sharing

Regards
Tom

jessegrunert, 2021-03-25:
No problem at all and thank you for the amazing product!