tomvandeneede/p2pp #102: Filament Presets Missing

destinybonavita, 2021-03-30:
Hi, when trying to setup PrusaSlicer for P2PP support all of my filament presets vanish. I reached out to Prusa support and they're saying it could be a P2PP issue. I'm not sure how but I figure I'd reach out.

[image]
[image]

As soon as I toggle "Single Extruder Multi Material" the second screenshot happens. Everything worked fine yesterday until it didn't. I hadn't updated or changed any settings either.

tomvandeneede, 2021-03-30:
How did you do the setup?   Did you start from a working profile?   Did you do the setup manually?

destinybonavita, 2021-03-30:
> How did you do the setup? Did you start from a working profile? Did you do the setup manually?
> […](#)

I followed the PDF to do the setup, I also started from the system preset for the MK3S+. Attached is my profile 
[Original Prusa i3 MK3S & MK3S+ (P2PP).ini.zip](https://github.com/tomvandeneede/p2pp/files/6229325/Original.Prusa.i3.MK3S.MK3S%2B.P2PP.ini.zip)

That profile only has the bare minimum changed and it still has the issue

tomvandeneede, 2021-03-30:
> Toggling the MMU2 printer profile fixed the issue prusa3d/PrusaSlicer#6292 (comment) <https://github.com/prusa3d/PrusaSlicer/issues/6292#issuecomment-810307135>

thanks for pointing this out