tomvandeneede/p2pp #103: Single Color with Extruder 1 error

Maglin, 2021-04-01:
If you are doing a single color print and use Extruder One P2PP won't process it and says. 

`Generate processed GCode
This does not look like a multi-colour file. 
Processing time 0.85s
Generating GCODE file: S:\Desktop\AB-BN-28\Overhang test v1_0.3mm_15m.mcf.gcode

-------------------
Print Summary
-------------------

Number of splices: 0
Number of pings: 4
Total print length 1723.36mm

Tower Delta Range 0.00mm - 0.00mm

----------------------
Inputs/Materials used:
----------------------

---------------------
- SPLICE INFORMATION-
---------------------
Splice Offset = 40.00mm
Autoloading Offset = 0.00mm`

It won't select the extruder.  It does work with Extruder 2-4.  The resultant file is useless when Extruder One is selected.

tomvandeneede, 2021-04-01:
ok will look into this.   This is intentional.   I have not altered this after color subnoot features have been added.   Will look into this
I assume this is why you want to use it right?

Maglin, 2021-04-02:
Well, I don't see any other good way to do single color.  The multi spool option on the palette is just a PITA to use,  I've found just assigning an extruder in the file just makes it easier.  And since it works with 2-4 it would be very nice if it worked on extruder one.  Then the Default would work as well and at least for myself and not end up with a print failure and wasted filament.  For a frame of reference, my printer runs almost 24/7.  It's probably accumulated about 3-4 weeks total in the last two years of none print time, and most of my prints are a single color.  Thank you for looking into it.

closed 2022-02-05 by tomvandeneede

ryanlindsley, 2022-12-12:
I've tried to do a single color print and getting the same error. I was using slot 3 (since it had the color I wanted). How am I supposed to do a single color print correctly?

tomvandeneede, 2022-12-12:
Hy Ryan,

I believe we tried to achieve this but would need to dig deeper to see if we actually managed to get this working.
For a single color print, loding the filament directly in the printer is probably the safest way for now

The error you are getting is indicating there is something with the splice table.  Can you please send me the mcxi file through PM and I will have a look… but still not sure why you would use palette for single color paints :-)

Regards
Tom

ryanlindsley, 2022-12-15:
Hey thanks for the reply Tom! I’d love to send more details. I’m not sure how to PM you through GitHub...

The reason I’d like to use the palette for single color prints is basically for convenience. Let’s say I load it up with 8 of my favorite colors. It’d be nice to select a color for a print (like on extruder 5) in PrusaSlicer, and send it to the palette with P2PP just like multi-color prints. So far if I want to do a single color print, I’ve had to disconnect everything, unload the palette, move the spool I want back to the Prusa, and transfer my print file with the SD card.

From the couple prints I tried, it seems that when it’s about to cut the filament near the end of the print (to retract the excess back and let the print finish up) it thinks there’s a splice that’s supposed to happen. In P2PP it even says the file has 1 splice…which, it shouldn’t. Right?

tomvandeneede, 2022-12-16:
Best is through PM

See where you want to go… makes sense