tomvandeneede/p2pp #104: Error: expected str, bytes or os.PathLike object, not NoneType

malcinator, 2021-04-04:
Error: expected str, bytes or os.PathLike object, not NoneType

Seeing the above error when running on macOS. Version 6.02.29 does not display the error

Using latest PrusaSlicer on macOS & Palette+
Settings in Start gCode

;Palette Plus P2PP Configuration
;P2PP PRINTERPROFILE=50325050494e464f
;P2PP ACCESSORYMODE_MSF
;P2PP P+LOADINGOFFSET=24135
;P2PP P+PPM=30.2125
;P2PP MATERIAL_PLA_PLA_1.5_2.5_0
;P2PP SPLICEOFFSET=40
;P2PP CONFIGEND

[image]

tomvandeneede, 2021-04-04:
apologies for this on, stupid mistake on my behalf.  should eb ok in 7.0.1

malcinator, 2021-04-04:
No need to apologize at all. Just wanted to make you aware of the issue.

closed 2022-02-05 by tomvandeneede