tomvandeneede/p2pp #105: Move out of Range

cfelicio, 2021-04-10:
Hi There,

Trying to get the Palette+ working with P2PP. Here is a new issue when trying to print the mosaic keychain. Midway thru the print, I get the following error:

!! Move out of range: 31.049 -188.299 5.920 [1713.518]

Looking at the Gcode generated, I see this:

;TYPE:External perimeter
G1 F3600
G1 X154.578 Y186.078 E0.00565
G1 X154.308 Y186.006 E0.00914
G1 X154.047 Y185.849 E0.00999
G1 X153.845 Y185.622 E0.00990
G1 X153.721 Y185.363 E0.00939
G1 X138.535 Y139.106 E1.59249
**G1 X31.049 Y-188.299 Z5.920 E0.19710**
; -------------------------------------
; --- P2PP - ACCESSORY MODE PING PART 2
;PING PAUSE 2 START
G4 P4000
G1
G4 P3000
G1
;PING PAUSE 2 END

; -------------------------------------
G1 X138.535 Y139.106 E1.39530
G1 X138.408 Y138.786 E0.01124

Please see attached all relevant files. Let me know your thoughts! :-)

[201901MosaicKeychain2.msf.zip](https://github.com/tomvandeneede/p2pp/files/6290292/201901MosaicKeychain2.msf.zip)

tomvandeneede, 2021-04-10:
Carlos,

which printer are you using?

cfelicio, 2021-04-10:
Hi Tom,

The Dbot MM, like this:

[image]

tomvandeneede, 2021-04-10:
can you get me the unprocessed print (the one without P2PP having processed it)

would ike to see what is in there

cfelicio, 2021-04-10:
Sure, here is with the P2PP commented out, as well as enabled.
[latest.zip](https://github.com/tomvandeneede/p2pp/files/6290583/latest.zip)

Just for fun, I removed the offending line on the gcode, and looks like got good results :)

[image]

tomvandeneede, 2021-04-10:
Hi,

I see you sent 2 STL but really would like to see the unprocessed gcode to make sure where the offending move comes from
print looks fine for the rest
not much you can improve

cfelicio, 2021-04-10:
Hi Tom,

 The file I sent on the last coment (latest.zip) should contain 2 gcode files, plus the msf file). The unprocessed file is called 201901MosaicKeychain2.unprocessed.gcode

I'm attaching it here as well as a new zip
[201901MosaicKeychain2.unprocessed.zip](https://github.com/tomvandeneede/p2pp/files/6290629/201901MosaicKeychain2.unprocessed.zip)

tomvandeneede, 2021-04-10:
looksd it over and all appears normal.  I cannot find the offending line though

cfelicio, 2021-04-10:
Its on line 21093, but only present on the file that goes thru p2pp. the prusa slicer output does not have the line.

see the p2pp file below

[201901MosaicKeychain2.p2pp.zip](https://github.com/tomvandeneede/p2pp/files/6290736/201901MosaicKeychain2.p2pp.zip)

tomvandeneede, 2021-04-10:
ok found it

will check

tomvandeneede, 2021-04-10:
uploading 7.0.5 that should fix the issue

cfelicio, 2021-04-10:
Thanks so  much! Confirmed fixed!

closed 2021-04-10 by cfelicio