tomvandeneede/p2pp #106: User Error: Palette+ Clarification

cfelicio, 2021-04-10:
I noticed that while printing with P+ Accessory mode, the mosaic keychain never transitioned:

Expected:
[image]

Actual:
[image]

Comparing the MSF from P2PP with Chroma, I noticed the PPM value was different. Turns out I used on my Gcode the following:

[image]

;P2PP P+PPM=108925
;P2PP P+LOADINGOFFSET=29631

When in reality, I should be using the PPM obtained from Chroma:

[image]

I made some changes to the doc to clarify this on the P+ section. Not sure how to check in, but leaving it here for reference in case other users make the same mistake, as its an easy one to overlook.
[P2PP user manual.docx](https://github.com/tomvandeneede/p2pp/files/6290410/P2PP.user.manual.docx)

tomvandeneede, 2021-04-10:
thanks for the update, I will upload the changes into the manual

cfelicio, 2021-04-10:
Awesome! Appreciate it!

closed 2021-04-10 by cfelicio

cfelicio, 2021-04-10:
Oh, just a quick note, noticed the code on the document still shows:

;P2PP ACCESSORYMODE_MAF
;P2PP P+PPM=<ppm value>
‘P2PP P+LOADINGOFFSET=<loading offset>

I think for the P+, correct is (fix the ' for ;, and MAF for MSF):

;P2PP ACCESSORYMODE_MSF
;P2PP P+PPM=<ppm value>
;P2PP P+LOADINGOFFSET=<loading offset>

Cheers,
Carlos