tomvandeneede/p2pp #107: Unknown command:"M0" / Unknown command:"T0" on Klipper

cfelicio, 2021-04-16:
Hi Tom,

With my other palette (the P2), I'm trying to use it in connected mode with Klipper, and it seems to work well. There is just one little snag on the gcode. I see that in the past you created the flag to account for this:

;P2PP KLIPPER_TOOLCHANGE

But I still see a couple of instances of T0 / M0 on the gcode, and Klipper complains about it:

21:39:09
// Unknown command:"M0"
21:39:09
// Unknown command:"T0"

These seem to show up very early on the gcode, and only happen 2-3 times. See example below:

[1_Black.p2ppk.zip](https://github.com/tomvandeneede/p2pp/files/6322613/1_Black.p2ppk.zip)

Appreciate your input, and let me know if you need any more info!

tomvandeneede, 2021-04-16:
Hi Carlos,

Created 7.0.7 that does no longer generate the T0 if you are using the KLIPPER setting.

The M0 I believe is needed to the P2 at the start of the print.  I don’t believe you can leave it out… you can try 
;P2PP DO_NOT_GENERATE_M0

it should stop the generation of the M0 command as you asked but the print may fail…..  I really hope this is just a warning and not the print stopping from this, in that case, you should now get one less warning.

Hope this gives an answer to your problems….

regards
Tom Van den Eede

cfelicio, 2021-04-16:
Perfect, thanks Tom!

closed 2021-04-16 by cfelicio