tomvandeneede/p2pp #108: Error: string index out of range with SuperSlicer 2.3.56

ashleyuk90, 2021-06-22:
Trying to use P2PP with SuperSlicer 2.3.56 getting "Error: string index out of range"

[SOURCE_SPLIT_HEAD_0.12mm_PLA_ENDER3V2_9h22m.mcf.zip](https://github.com/tomvandeneede/p2pp/files/6693742/SOURCE_SPLIT_HEAD_0.12mm_PLA_ENDER3V2_9h22m.mcf.zip)

-----------
;P2PP PRINTERPROFILE=xxxxxxxxxxxxxxxxx
;P2PP SPLICEOFFSET=40
;P2PP MINSTARTSPLICE=100
;P2PP MINSPLICE=70
;P2PP EXTRAENDFILAMENT=150
;P2PP LINEARPINGLENGTH=350
;P2PP ABSOLUTEEXTRUDER
;P2PP KLIPPER_TOOLCHANGE
-----------

P2PP Version: 7.01.11
Python Version: 3.9.1
Reading File C:\Users\ashle\Desktop\SOURCE_SPLIT_HEAD_0.12mm_PLA_ENDER3V2_9h22m.mcf.gcode
Analyzing Prusa Slicer Configuration
All purge lenghts 70/70 OR 140. Purge lenghts may not have been set correctly.
SPLICE OFFSET: 40.00mm
Extra filament at end of print 150.00mm
Linear Ping interval of 350.00mm
Convert to absolute extrusion parameters

We're sorry but an unexpected error occurred while processing your file
Please sumbit an issue report on https://github.com/tomvandeneede/p2pp

Error: string index out of range

Traceback Info:
File "p2pp.py", line 88, in 
File "C:\Users\tomvandeneede\Documents\p2pp\p2pp-dev_Qt\test\_build_update_\p2pp\p2pp\mcf.py", line 737, in generate 
File "C:\Users\tomvandeneede\Documents\p2pp\p2pp-dev_Qt\test\_build_update_\p2pp\p2pp\psconfig.py", line 209, in parse_prusaslicer_config

tomvandeneede, 2021-06-22:
Hey,

it seems it fails to process a config line:

; extrusion_width = 

the extrusion width does not seem to have been specified.
Can you please check your config?

thanks

ashleyuk90, 2021-06-22:
It is defiantly set, however by default SS has it set in % where PS has it in mm

[image]

Maybe its failing at the fact the value isnt what it's expecting?

JVdbKtr, 2021-09-06:
I was looking into a similar setup with SuperSlicer 2.3.56.7 and ran into this exact issue.
Upon inspection of the SuperSlicer output file extrusion_width is simply not set - there is no value behind "; extrusion_width = " and thus this does not appear to be a P2PP issue but rather a SuperSlicer issue.

tomvandeneede, 2021-09-06:
Hi,

 

Would you mind sharing the unprocessed gcode file:   Will have a look but
need some further test background

You can PM me on Facebook if you want

regards

SuperSlicer 2.3.56 (#108)

 

<https://github.com/tomvandeneede/p2pp/issues/108#issuecomment-913806875> ,
UDAXANCNFSM47DOX5XA> .
t=8&pt=524675>  or Android
tm_campaign%3Dnotification-email%26utm_medium%3Demail%26utm_source%3Dgithub>
<https://github.com/notifications/beacon/AK6CHJNLZJFWO7BPRJGWITLUAUDAXA5CNFS
M47DOX5XKYY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOGZ3ZMGY
.gif>

JVdbKtr, 2021-09-07:
Afer some Facebook discussion with @tomvandeneede we switched the default extrusion width value to an absolute value (not a % but the actual mm) at which point p2pp was happy again.
Interestingly after switching back to the original % value it still worked, so it's almost as if SuperSlicer needs to see a change to the values before adding the variable to the output file.

Thanks to Tom for the quick responses!

closed 2022-02-05 by tomvandeneede