tomvandeneede/p2pp #110: Palette 3 and Klipper

audiorazor, 2021-09-24:
Do you know if the new palette will work with p2pp?

tomvandeneede, 2021-09-24:
Hi
tjis is a question best asked in the Mosaic groups on fb.  I don’t have insight on the controller but given it is an extension of standard gcode I don’t think it will  be an issue

closed 2021-10-06 by audiorazor

Yezariael75, 2023-03-02:
Would like to bring this up, is there any chance for Klipper? Thanks!

tomvandeneede, 2023-03-02:
Currently only accessory mode, working on direct mode but on hold pending p3 firmware upgrade

ThomasNyk, 2023-11-22:
Any update on this? What are we waiting on? Mosaic, klipper or a third entity?

tomvandeneede, 2023-11-22:
Waiting for Mosaic to allow remote pings through MQTT
it is pending work with lower priority on their side  since they are workington the release of the Element and Array devices
Hopefully they will get the update out soon after the release
should not be a lot of work I would guess