tomvandeneede/p2pp #115: Error: string index out of range

emersonsc, 2021-11-03:
When trying to export any file using SuperSlicer, I get the following error:

P2PP Version: 7.02.00
Python Version: 3.9.1
Reading File A:\Emersons Stuff\Printer Stuff\Test\201901MosaicKeychain1_0.2mm_PETG_ENDER3BLTOUCH_57m.gcode
Analyzing Prusa Slicer Configuration

We're sorry but an unexpected error occurred while processing your file
Please sumbit an issue report on https://github.com/tomvandeneede/p2pp

Error: string index out of range

Traceback Info:
File "p2pp.py", line 88, in 
File "C:\Users\tomvandeneede\Documents\p2pp\p2pp-dev_Qt\test\_build_update_\p2pp\p2pp\mcf.py", line 737, in generate 
File "C:\Users\tomvandeneede\Documents\p2pp\p2pp-dev_Qt\test\_build_update_\p2pp\p2pp\psconfig.py", line 209, in parse_prusaslicer_config

tomvandeneede, 2021-11-04:
7.02.2 or later should fix this issue
when running PS2.4  make sure to rub 1.00.xxx as 7.xxx version only supports up to PS2.3

emersonsc, 2021-11-04:
Where can I get that version because I used the download link from github
and that's the version I got

emersonsc, 2021-11-04:
> 7.02.2 or later should fix this issue when running PS2.4 make sure to rub 1.00.xxx as 7.xxx version only supports up to PS2.3

7.02.3 fixes the issue with Prusa 2.4, but 1.00.205 does NOT fix this issue with SuperSlicer, I get the same error just worded slightly different

P2PP Version: 1.00.205
Python Version: 3.9.1
Reading File A:\Emersons Stuff\Printer Stuff\Test\J&K Props.gcode
Analyzing Prusa Slicer Configuration
Extrusionwidth set to 0, defaulted back to 0.45
Analyzing Layers / Functional blocks
Tower detected from (79.55499999999999, 166.97) to (146.93699999999998, 200.152)
Bed origin (3.0mm, -2.0mm)
Bed size (225.0mm, 230.0mm)
P2PP Layer Configuration is missing!!

closed 2021-11-04 by emersonsc

tomvandeneede, 2021-11-04:
then this is likely an incompatibility between prusaslicer and super slicer.
Can you please send me the unprocessed gcode and I will have a look at it. will try to add for a later release