tomvandeneede/p2pp #116: LAYER configuration is missing

emersonsc, 2021-11-03:
When trying to export any file in PrusaSlicer, i get the following error:

P2PP Version: 7.02.00
Python Version: 3.9.1
Reading File A:\Emersons Stuff\Printer Stuff\GCode\Hitch Covers\201901MosaicKeychain1_0.2mm_PETG_ENDER3_55m.gcode
Analyzing Prusa Slicer Configuration
Analyzing Layers / Functional blocks
Bed origin (0.0mm, -5.0mm)
Bed zise (220.0mm, 225.0mm)
LAYER configuration is missing.

emersonsc, 2021-11-04:
> 7.02.2 or later should fix this issue when running PS2.4 make sure to rub 1.00.xxx as 7.xxx version only supports up to PS2.3

7.02.3 fixes the issue with Prusa 2.4, but 1.00.205 does NOT fix this issue with SuperSlicer, I get the same error just worded slightly different

P2PP Version: 1.00.205
Python Version: 3.9.1
Reading File A:\Emersons Stuff\Printer Stuff\Test\J&K Props.gcode
Analyzing Prusa Slicer Configuration
Extrusionwidth set to 0, defaulted back to 0.45
Analyzing Layers / Functional blocks
Tower detected from (79.55499999999999, 166.97) to (146.93699999999998, 200.152)
Bed origin (3.0mm, -2.0mm)
Bed size (225.0mm, 230.0mm)
P2PP Layer Configuration is missing!!

tomvandeneede, 2021-11-04:
Not sure the Layer config is something that is needed and unchanged since almost one year… 7.02.00 or 7.02.02 need the same layerconfig as was in the screenshot from the installation manual
if this is insufficient, this is still al Superslicer issue and will need to look at the gcode…. though  I would be ssurprised as I have other users working with super slicer

emersonsc, 2021-11-04:
Here you go, hope this all helps

[Files.zip](https://github.com/tomvandeneede/p2pp/files/7478842/Files.zip)

> Not sure the Layer config is something that is needed and unchanged since almost one year… 7.02.00 or 7.02.02 need the same layerconfig as was in the screenshot from the installation manual if this [...]

tomvandeneede, 2021-11-06:
Please have a look at the installation instructions.  I still believe your Layer configuration is incorrect!  It is missing the gocde that needs to be pasted into the after layer change gcode.

closed 2022-02-05 by tomvandeneede