tomvandeneede/p2pp #118: Complete With Warnings.

gbrspratt, 2022-01-03:
Sorry if this is the wrong place.... but I don't use Facebook anymore otherwise id ask on there!

I'm having issues with the new version. I previously followed a tutorial to get it all setup on my old printers and it worked flawlessly. I've just built a voron 2.4. I can use canvas but I'm working on the external buffer to be able to pump the speed up and for that I need p2pp because I can use linear pings. So I've removed the older versions of p2pp(or so I thought?) and downloaded the latest windows zip file. followed the instructions, set it all up and loaded a test model. here's where it all starts going wrong. First of all it says its version 4.16.000. I thought I downloaded 8.00.17 (I've also tried 8.00.16)
[image]

when I come to export the sliced gcode I get a warning that it priming will not work p2pp. Then..... it says it doesnt look like a multi material print? it looks fine in superslicer with 100 tool changes but none of them come through to p2pp.

[image]

I'm hopeful its just because i have an old version lurking around on my laptop somewhere but i can see it...... and if its not that then hopefully its just a silly mistake somewhere I've overlooked.

Matt

tomvandeneede, 2022-01-03:
Hi 

just install 8.0.17 according to the wiki pages on GitHub.   First remove the 4.0.16 version (should be the p2pp folder on your desktop)
then go to the location where you unzipped 8.0.17 and double click the executable to get the location to copy into PS2.4. output post script processing routine

then you should be good.

Let’s take things from that point before solving processing issues…

gbrspratt, 2022-01-03:
I started again, and now definitely have the correct version :)
[image]
But some weird things are now happening. I'm getting some weird purge in the front left corner. This is obviously what p2pp was referring to about purge? and its still not realising it is multi material.

thanks for the quick response as usual. Is there a way to donate to the project to show a little bit of support and appreciation for you?

Matt

tomvandeneede, 2022-01-03:
This is weird.  Would you mind sending met the gcode file then I can take a quick look.

gbrspratt, 2022-01-03:
[cali_1-R.mcf.zip](https://github.com/tomvandeneede/p2pp/files/7803242/cali_1-R.mcf.zip)
hopefully that works.

tomvandeneede, 2022-01-03:
try adding the following line to your startup gcode

;P2PP KLIPPER_TOOLCHANGE

gbrspratt, 2022-01-09:
Thanks Tom! Sorry it took so long. That has sorted it and SuperSlicer had a little tick box hidden away "prime all printing extruders" that I missed on setup. I haven't run a test print yet, but P2PP seems to be doing its part. 

Thanks again

closed 2022-01-09 by gbrspratt

tomvandeneede, 2022-01-09:
thanks for confirming!