tomvandeneede/p2pp #119: Purge tower got speed up drastically when processed

Siwat2545, 2022-02-03:
The speed of the purge tower increases drastically and also have some inconsistent speed
Before Process:
[image]
After Process:
[image]

Project file, process, unprocessed gcode file attache
[bugreport.zip](https://github.com/tomvandeneede/p2pp/files/7993929/bugreport.zip)
d

tomvandeneede, 2022-02-03:
Will look into this

tomvandeneede, 2022-02-04:
Hi I think I found what is causing but neet some further testing before compiling ut a new version of the tool.   Hope to get around this during the weekend.

tomvandeneede, 2022-02-05:
Can you please check with 8.0.17 as it seems to process fine on my latest version
[image]
There was an issue in your gcode for sure with the high speed, but when processing with the latest 8.0.17 version the speeds seem to be what is about to be expected.

Siwat2545, 2022-02-05:
The newest version does indeed fix the issue, thank you.
This also raise another issue, your update checking seems to not be working. (:

closed 2022-02-05 by Siwat2545