tomvandeneede/p2pp #120: Swapping colours manually

XylenC4, 2022-02-13:
I would like to swap colours without the mentioned plugin as I'm using KLIPPER Firmware with MainsnailOS.

I've noticed the source does contain a function called "MANUAL_SWAP" but nothing happens on generation when I am enabling it.
I'll expect M25 inside my GCODE, but it's not in there.

Furthermore, I've noticed this function is marked as "Not Documented" but I'd like to give it a try.
Which requirements must be met to use this function?

[Generation.txt](https://github.com/tomvandeneede/p2pp/files/8056824/Generation.txt)
[SVG_0_0.txt](https://github.com/tomvandeneede/p2pp/files/8056825/SVG_0_0.txt)

tomvandeneede, 2022-02-14:
MANUAL_SWAP was written for a specific printer and it is not used/maintained for general use
It is hardcoded around the Y-commands and does not support the Tool change commands in KLIPPER
It also only works in standard mode not in any of the special modes like full-urge etc

It really is mor e of a test feature for very specific hardware hence not documented/supported.

With the new v2.4 GCODE process in Prusaslicer itself you may be able to joy replace all toolchange commands with a G25 easily

XylenC4, 2022-02-14:
Thanks for your replay, maybe get back a step.

There are only two colour swaps on the first layer, so there is no need for me to get an extra message on octoprint.
It would be enough to get a response from the Palette to change out the filament, but unfortunately there is no one.
When changing out the filament, my print will stop where it is (of course, I could put it manually in parking position).

If the 5. Material is required by the Palette the outgoing gear spins, it seems to be a bug as the required material is not available.

My gcode does not contain any O500/O501 commands. Are these commands generated by the Plugin?
As I understand the PP2P documentation, the plugin does only provide early messages on the octoprint page.

tomvandeneede, 2022-02-14:
> Thanks for your reply, maybe get back a step.

O500 is an old test function to print more than 5 colors on the P2
It is no longer supported however and never has been fully tested as it required a none-standard plugin n the OctoPrint side.
technically I could look into doing M25 and allow more than 4/8 colors by pausing the print and sending a message to swap filament but it is not very easy.   It requires a lot of extra parameters to be set to make sure the pause does not ruin your print (e.g hover over the tower and not the perimeter of the print etc…. hence it needs to know the historical modifier, the loading length etc.

a lot of fun but turns out to get complex very quickly… but all of course within the realm of P2/P3 so no need to physically swap out filament

XylenC4, 2022-02-14:
Thanks for your input.

I will play a bit with python, maybe I can make a "proof of concept" post processor to add M25.

The main limitation I think is the length of the tube which has to be known.
The print must run before using M25 the first time so on the set-up phase of the palette the color is not allowed.

Changing filament could be made two color slices before (to medigate the historical modifier) on a purge tower (maybe cool down the hotend until the user reacts)

tomvandeneede, 2022-02-15:
What exactly do you want to do
which hardware do you have?
If you you replace any T-command (except maybe the first) with M25 you should be fine
you additionally may want to remove the surplus code for swapping filament (which is also fairly easily done)
Just the first can be done with the new functionality to edit code from within Prusa Slicer
The latter might be possible but likely a little bit more tedious

XylenC4, 2022-02-19:
Forcing the printer to stop at the wipe tower to change out the filament. But I've been not successfully to change the filament using 5 colours and inserting M25 manually. The palette will not allow unloading any filaments.

I'll assume the best would be to switch over to octoprint and using the official plugin.