tomvandeneede/p2pp #122: Model results in incorrect filament preparation (incorrect feeding length, unnecessary cut of filament, incorrect initial filament length)

ServerOnTheMove, 2022-02-25:
**Versions**
PrusaSlicer 2.4.1-beta1
MacOS Monterey 12.2.1 (Intel)
Prusa i3 MKS3+ Firmware 3.10.1, no modifications
Mosaic Palette 3 Pro - Firmware alt-22.02.09.0

**Issue**
I created a design and to test the design I only need the first couple of layers (only the first layer contains colour changes) I created a test version that only prints the first 4-5 layers.

P3P starts fine and generates the initial splice length. When it comes to filament loading the issue starts:

There is no filament coming out of the PFTE tube (on the printer end) to feed into the extruder gear
The P3P screen shows over 200mm length to feed, where it is typically only approx 70mm
If I remove the PFTE from the Palette and pull out the filament (very slowly until the inner sensor is triggered), I can see the feed counter on the P3P decreasing
As soon as the sensor is triggered, the P3P cuts the filament (colour red) and pushes it through the splicer core without splicing it together resulting in a cut and separate filament.
See video here https://1drv.ms/u/s!Ap87vvDWL0D_hZcMjljSsDUM8MEV4w?e=cm8TbU

**Steps needed to reproduce the problem**
I don't know if this can be reproduced with any model.
I took a model and cut it down to the first 3-4 layers.
To reproduce I can share the mcfx and the 3mf file for a version with the issue (matching the video) and a version where it works

Is this a new feature request?
No

Project File (.3MF) where the problem occurs
Please contact me directly to get the files - I don't want to share them here to the world.

The video of the issue
https://1drv.ms/u/s!Ap87vvDWL0D_hZcMjljSsDUM8MEV4w?e=cm8TbU

tomvandeneede, 2022-02-25:
Can you send me the gcode file pleas e

would like to take a look at this in the code

any chance the total amount of filament needed for the print is less than the feeding tube length?   In that case you will need to increase the extra filament length parameter

ServerOnTheMove, 2022-02-25:
Hi Tom,
thanks for getting back to me.

If I respond to the github email address - will this go directly into your
inbox or will it go onto the issue in github?

Thanks,
Dante

ServerOnTheMove, 2022-02-25:
Hi Tom,
can you please send me an email to dante@serveronthemove.com.au so that I can send you the files directly?
Not keen to share them here - sorry.

Thanks,
Dante

tomvandeneede, 2022-02-25:
will go on GitHub, you can PM me in facebook messenger for private discussion

> On 25 Feb 2022, at 07:08, Server On The Move ***@***.***> wrote:
> 
> nbox or will it go onto the issue in github

ServerOnTheMove, 2022-02-25:
Tom, I sent you a PM on the facebook machine.