tomvandeneede/p2pp #123: Purge Tower has lines with Layer tall layer height then drops down to first layer when filling

Boostang99, 2022-02-26:
I am just starting to use P2PP with my setup but am running into a issue. When the purge tower is created it does 5 very spaced out lines after the first layer is laid. These 5 lines get built upon with 5 layers up to a height of 0.95mm. Then at 0.95mm the print needs color change. The purge tower is then attempted to be bridged to infill. However the new print height selected for bridging is all the way back down to 0.15 and the printer is trying to smush the layers down which causes the filament not to be extruded due to pressure and the direct drive starts to skip. I do not believe that this is correct but cannot figure out how to fix it. Originally in my G-code I was using
;P2PP FULLPURGEREDUCTION
;P2PP WIPEFEEDRATE=3000
;P2PP AUTOADDPURGE

I then switched it to Purge Tower Delta with a variance of 999, still same thing happening

I then changed the variance to 0 to disable Delta feature. Still same results

I then erased the feature all together and still running the G-code the same. 

Any help would be greatly appreciated. I am unsure where my custom G-Code went wrong. My printer is a Prusa MK3s+ and I am using the PP3

Here is my current G-Code configuration. Thank you again.

M862.3 P "[printer_model]" ; printer model check
M862.1 P[nozzle_diameter] ; nozzle diameter check
M115 U3.10.1 ; tell printer latest fw version
G90 ; use absolute coordinates
M83 ; extruder relative mode
M104 S[first_layer_temperature] ; set extruder temp
M140 S[first_layer_bed_temperature] ; set bed temp
M190 S[first_layer_bed_temperature] ; wait for bed temp
M109 S[first_layer_temperature] ; wait for extruder temp
G28 W ; home all without mesh bed level
G80 ; mesh bed leveling
G1 Z0.2 F720
G1 Y-3 F1000 ; go outside print area
G92 E0
G1 X60 E9 F1000 ; intro line
G1 X100 E12.5 F1000 ; intro line
G92 E0
M221 S{if layer_height<0.075}100{else}95{endif}

; Don't change E values below. Excessive value can damage the printer.
{if print_settings_id=~/.*(DETAIL @MK3|QUALITY @MK3).*/}M907 E430 ; set extruder motor current{endif}
{if print_settings_id=~/.*(SPEED @MK3|DRAFT @MK3).*/}M907 E538 ; set extruder motor current{endif}

;P2PP PALETTE3_PRO
;P2PP PRINTERPROFILE=12345678901234567890123456789012

;P2PP SPLICEOFFSET=40
;P2PP MINSTARTSPLICE=130
;P2PP MINSPLICE=90
;P2PP EXTRAENDFILAMENT=150
;P2PP LINEARPINGLENGTH=350

;P2PP MATERIAL_PLA_PLA_1_0_1
;P2PP MATERIAL_DEFAULT_1_0_1

;P2PP FULLPURGEREDUCTION
;P2PP WIPEFEEDRATE=3000
;P2PP AUTOADDPURGE

;P2PP CHECKVERSION

[image]

tomvandeneede, 2022-02-26:
which version of P2PP are you using?

tomvandeneede, 2022-02-26:
Could you please also PM me the unprocessed gcode so I can run the processing in the debugger to take a look

Boostang99, 2022-02-26:
Thank you very much for your help. I am currently running version 8.00.19 

I have also sent the g-code file as well. Thank you again!

tomvandeneede, 2022-02-26:
where did you send the gcode file?

> On 26 Feb 2022, at 14:24, Boostang99 ***@***.***> wrote:
> 
> ave also sent the g-code file as well. Thank yo

Boostang99, 2022-02-26:
Sorry I sent this message with the intent on sending the g-code right after. However I am a noob with Github and cannot figure out how to send a message... Where do you want me to send?

tomvandeneede, 2022-02-26:
Hi,

would need the unprocessed gcode (gcode without running P2PP)
best it to PM to my facebook account though if not too big you should be able to attach here as well.

regards
Tom

Boostang99, 2022-02-26:
I have sent the file to your facebook. Thank you very much again

tomvandeneede, 2022-02-27:
issues addressed in 8.0.20

closed 2022-02-27 by tomvandeneede