tomvandeneede/p2pp #126: Ping Generation Accessory Mode

superblobmonster, 2022-03-13:
**P2PP Script Output**
Version 30
[image]

GCODE generated by SuperSlicer_2.3.57 for a Klipper machine in accessory mode. Model is the Mosaic calibration key-chain
[P2PP_Keychain_V30.zip](https://github.com/tomvandeneede/p2pp/files/8239005/P2PP_Keychain_V30.zip)

**Issue**
Pings are not being generated at expected interval while over purge tower.

It appears to be an issue where the pings only occur when there is a tool change. On models with layer changes on every layer, the pings appear to return to what appears to be every layer, or at least a regular interval. I have not tested this in a print.
[P2PP_Keychain_qty2_V30.zip](https://github.com/tomvandeneede/p2pp/files/8239015/P2PP_Keychain_qty2_V30.zip)

As and aside, in the P2PP output, the colors of the [###] is correct, but the description is slightly inaccurate!

tomvandeneede, 2022-03-13:
8.0.31 should solve this issue

superblobmonster, 2022-03-13:
Yep!

Calibration print was successful, running larger print now!
Got one ping that started the first pause on the transition tower, and the second pause over the part.

Thank you Tom!

superblobmonster, 2022-04-04:
Continuation of Accessory pings-
If a ping starts late in the purge tower, the second pause can occur off the purge tower and on the main print. 
Pings seem to have a chance of being missed. Worth noting, I have a long ~120mm output tube which may be 'smearing' the ping pause.

tomvandeneede, 2022-04-04:
Ben,

that is a choice to be made.   It is very difficult to get all pings over the tower at this moment with the current processing.   will see is anything can be improved but it will not be so easy
I do not understand why this would contribute to pings being missed,   The pings are 2 pauses within between a defined extrusion, unless the extrusion in-between suffers from the long tube (i.e. through compression of generated filament) there should be no reason for this pings to go undetected.   Anyway there is not much that can be done from gcode perspective to solve this I think

closed 2022-04-04 by superblobmonster