tomvandeneede/p2pp #129: Pings finishing over model

superblobmonster, 2022-04-08:
Known problem, just creating a github issue to track it.

V33 compiled from source on Linux, accessory mode, klipper tool changes, P3P
On rare occasions pings can be generated where the second pause in a ping is placed over the model instead of the purge tower.
In even rarer occasions it will be printing a small feature and land on an external perimeter which creates a substantial blob on the surface finish.

[image]

[3C_Lizard.zip](https://github.com/tomvandeneede/p2pp/files/8454918/3C_Lizard.zip)

tomvandeneede, 2022-04-09:
In theory a very easy problem to solve but will require a full rewrite of the code to get this resolved. This will take a substantial rewrite of P2PP to resolve.  Looking into alternative ways to handle this

tomvandeneede, 2022-04-09:
Added a 3mm retract (currently there was none) to as a temp countermeasure

superblobmonster, 2022-04-29:
Would it be possible to also add a 1mm Z-hop to prevent the nozzle from melting a hole into the model? I recognize not all printers have a fast z-hop, maybe a variable z-hop setting?

tomvandeneede, 2022-04-30:
Hi Ben,

this is a very generic question…  Do you mean adding a Z_Hop when doing thethe pause 9like 1cm or so) to keep the heat away from the model?  or do you mean w Z-hop when moving..  not entirely clear to me

superblobmonster, 2022-04-30:
Thanks Tom,
I just mean lifting the tool head during the inserted ping to keep heat away from the model. The slicer settings take care of Z-hop during travel moves.

tomvandeneede, 2022-04-30:
need to do some thinking about that as it adds to the pause and will most likely fail to detect the extrusion… a short move (3mm) for the retract may not be enough to move the filament all the way back into the P3 to detect movement
Will be a trial thing for sure...