tomvandeneede/p2pp #130: Purge speed settings persistant

superblobmonster, 2022-04-09:
While investigating purge tower speeds, I set the PURGETOSPEED parameter for P2PP, which worked as intended.

With this setting now removed, it appears as though the speed setting is still used until a ping occurs, which then returns to the expected speed from the original gcode.
[image]

Accessory mode, klipper tool changes, P3P, running V34 on Linux running from source
[Snail.zip](https://github.com/tomvandeneede/p2pp/files/8456862/Snail.zip)

tomvandeneede, 2022-04-09:
Really not sure what you mean by this….

superblobmonster, 2022-04-09:
The output of SS does not change speed during the purge, my start g-code does not have PURGETOSPEED set anymore
But the output of P2PP still sets the last PURGETOSPEED for part of the purge tower until a ping occurs, where it then slows back down.
I have no clue what would cause this, but it seems to me like the last used PURGETOSPEED setting is getting saved somewhere and re-added to the output. Or perhaps this is some other normal/expected behavior of P2PP that I'm not familiar with

Purge tower speed as output by SS:
[image]

superblobmonster, 2022-04-29:
Checked V41,
still very strange speed behavior exporting from SS. Purge tower is bouncing between two speeds, and the first skirt perimeter is way too fast.

SS preview of skirt and purge tower on left, P2PP gcode preview on right. Same results with and without PURGETOSPEED
[image]

tomvandeneede, 2022-04-30:
Ben,

can you please provide me with processed and unprocessed gcode files where I can test/check this with

regards

superblobmonster, 2022-04-30:
[Wings_Raw.zip](https://github.com/tomvandeneede/p2pp/files/8596911/Wings_Raw.zip)

From the facebook group I've also realized it's "PURGETO**P**SPEED" not "PURGETOSPEED"

qkpony, 2022-07-21:
That's crazy because i was going to ask about this also. I also use Klipper and Palette in accessory mode. But when it gets to the purge tower the speed will suddenly go very fast and usually mess up my purge tower and cause all kinds of problems. The purge tower speed will start normal then all of a sudden it speeds way up.

tomvandeneede, 2022-10-11:
PURGETOPSPEED is defaulted to 3000mm/min so even if the parameter is not set, that value is applied.   If you want it higher, you need to set the parameter to the desired value

superblobmonster, 2022-10-30:
New updates, I'm getting even stranger behaviors.
Purge tower speed is still not being adjusted by P2PP, or at least not in a predictable way.

**System Info:**
P2PP Version 49, downloaded Saturday, compiled from source on Ubuntu 22.04
SuperSlicer_2.4.58.5_2022-10-27 compiled from source
Printer is a Voron 2.4 running klipper,
Palette is a P3P in accessory mode.

**File:**
[Lizard_Multicolor.zip](https://github.com/tomvandeneede/p2pp/files/9896320/Lizard_Multicolor.zip)

Desired behavior: Purge tower speed control, > 100mm/s

**Points of note:**
1. Line 123541 post-processed, strange ~120mm/s single extrusion
2. Line 138784 post-processed, Same isolated high speed single extrusion.
3. Line 291047 post-processed, line 289528 unprocessed, purge line omitted
4. Line 292988 post-processed, line 291460 unprocessed, purge line omitted
5. Layer 58 is also the only full purge layer with speed modification, and it is still only ~50mm/s.

I have an open question with the Super Slicer discord as to what the purpose of the omitted purge lines are, they only seem to appear on the first layer (Line 9218 post-processed), and layers that bridge over supports in the purge tower, unless it is the last layer of the purge tower. Very strange. It's definitely a distinct line from the 'normal' purge lines, and I have no idea why it is generated by Super Slicer.

Worth noting I have convinced myself that first purge line is causing the first ping to be missed due to a long slow travel move in-between the first and second pause in pings. If there's a way to only ping after this mystery purge line I think that would fix the issue.