tomvandeneede/p2pp #131: Prusaslicer locking up and crashing during slicing (Initial setup)

thetechy2, 2022-04-26:
I load the model and hit slice and it get to like 85% and says generating supports and stays there. If i try slicing without supports it just locks up at 70% and does much the same.  after doing this nothing appears on the buildplate in layer visualization until i switch to model render in the bottom left and switch it back, all the while still locked up at the same percentage. If/when i hit export gcode the little dialouge box will popup on the bottom right but nothing further will happen. If i try to stop the slicing on that percentage dialouge box it will hard lock and crash. or try and change any other settings pretty much anywhere else in the program. Running prusa slicer 4.2.1 any help would be appreciated. thanks

tomvandeneede, 2022-04-26:
Canyon please provide some screenshots …. P2PP only comes into play when exporting the file so if something already goes wrong during the slicing this is a PrusaSlicer issue (most likely)

Do you have an old P2PP windows still open by change (you should not be able to export until the previous one is closed anyway, but maybe in your build of PS this behaved differently…

thetechy2, 2022-04-26:
[image]

thetechy2, 2022-04-26:
and when this is happening prusaslicer is also constanty eating more and more ram (15gb and counting) and has a hard hold on 25% of my cpu(9700k) so i agree something is probably wrong with my ps configuration, but the setup process is so convoluted like the guide tells you to jump all over the place between all the different settings tabs and gcodes. Im new to prusa slicer and it also sucks that all those settings are spread out as much as they are