tomvandeneede/p2pp #132: BB3B Blob creation not honoring absolute extrusion

unclej84, 2022-04-27:
Hello,

I noticed that the extuder always pulled back a lot of filament after trying to create the blobs. It seems that the E values within the `;---- BIGBRAIN3D SIDEWIPE BLOB` section are always relative values - I tried this with and without `;P2PP ABSOLUTEEXTRUDER` settings and this section is always the same:

`;P2PP BIGBRAIN3D_BLOBSIZE=30
;P2PP BIGBRAIN3D_BLOBSPEED=200
;P2PP BIGBRAIN3D_COOLINGTIME=12
;P2PP BIGBRAIN3D_PURGEPOSITION=261.45
;P2PP BIGBRAIN3D_FAN_OFF_PAUSE=2000
;P2PP BIGBRAIN3D_ENABLE
;P2PP BIGBRAIN3D_PRIME_BLOBS=4
;P2PP BIGBRAIN3D_NUMBER_OF_WHACKS=1
;P2PP BIGBRAIN3D_CLEARANCE_MM=0
;P2PP BIGBRAIN3D_RETRACT=0
;P2PP BIGBRAIN3D_SMARTFAN`

`;---- BIGBRAIN3D SIDEWIPE BLOB 2 -- purge 30.000mm
M107 ; Turn FAN OFF
G4 P2000 ; delay to let the fan spinn down
G1 X251.450 F3000 ; go near the edge of the print
G1 X261.450 F1000 ; go to the actual wiping position
G1 F200 E 7.500 ; Purge FAN OFF 
M106 S32 ; Set FAN Power
G1 F200 E 7.500 ; Purge FAN 12% 
M106 S64 ; Set FAN Power
G1 F200 E 7.500 ; Purge FAN 25% 
M106 S96 ; Set FAN Power
G1 F200 E 7.500 ; Purge FAN 37% 
G1 E0.00000 F1200
M106 S255 ; Set FAN Power
G4 S12 ; blob 12s cooling time
G1 X241.450 F10800 ; activate flicker
G4 S1 ; Mentally prep for second whack
G1 X251.450 F3000 ; approach for second whach
G1 X261.450 F1000 ; final position for whack and......
G1 X241.450 F10800 ; WHACKAAAAA!!!!`

tomvandeneede, 2022-04-28:
Which version of P2PP are you using and can you please include a processed and unprocessed gcode file.

unclej84, 2022-04-28:
Hello, it is latest version of plugin 8.0.39. The processed and unprocessed files are already attached as zip in the first post

tomvandeneede, 2022-04-28:
`SMARTFAN is actually an unsupported feature, but 8.0.40 should fix the behaviour with absolute extruders.   SMARTFAN is advices against by the people of BB3D because of the formation of the blob may be impaired.

unclej84, 2022-04-28:
Thanks for the quick fix - works

closed 2022-04-28 by unclej84