tomvandeneede/p2pp #133: 8.0.40: Blob count sets all splices to fixed length in combination with AUTOADDPURGE

unclej84, 2022-04-28:
I have tried to use blob count for BB3D instead of purge volume and get all length fixed to 130mm in the summary - also the total number of blobs is different even if I do not get a short splice on both cases.

Here are all files I got from this. Do I have to worry if using blob count?

tomvandeneede, 2022-04-28:
no files here.

unclej84, 2022-04-28:
sorry, upload took a while - now

tomvandeneede, 2022-04-28:
the automatic splice extension is not compatible with the blob count feature.  There is no plans to include that.  Added the filter so the length correction is not applied in the overview

unclej84, 2022-04-28:
Okay, good to know - so I will stick with the volumes then. Thanks for clarification.
Maybe it makes sense to add an elif [here](https://github.com/tomvandeneede/p2pp/blob/4bd466a02aeb48937eb9aa648d03608cf2f44f7f/p2pp/mcf.py#L931) too because it still shows that it would be activated. Maybe some hint like "Automatic Splice length increase cannot be used in conjunction with blob counts for BB3D".

Issue can be closed

closed 2022-04-28 by tomvandeneede