tomvandeneede/p2pp #135: [Feature Request] Splice settings info

Bmaenz, 2022-05-09:
I would love if P2PP showed me what all the splice settings are for each material that was used on the output screen. Just so I can easily verify that yes I did have the right splice settings for Material A going to Material B and so on.

Thanks!

Dandydorf, 2023-01-04:
I wish the P2PP team still worked on this. If prusaslicer newer versions were still usable, i would be able to use p2pp right this second.... 2.3 is totally broke

tomvandeneede, 2023-01-04:
This is still being worked on, have received little or no feedback on non-working issues lately, but addressed all of them as they passed.    PS2.5 roll-out was pretty uneventful with no major changes in terms of code generation.   Has some great features but they don’t seem to interact with P2PP..

Dandydorf, 2023-01-04:
> This is still being worked on, have received little or no feedback on non-working issues lately, but addressed all of them as they passed.    PS2.5 roll-out was pretty uneventful with no major changes in terms of code generation.   Has some great features but they don’t seem to interact with P2PP..  
> 
> > 
> > 
> > I wish the P2PP team still worked on this. If prusaslicer newer versions were still usable, i would be able to use p2pp right this second.... 2.3 is totally broke

Ahh okay i don't mean to nag on the build i just am  VERY frustrated with my bricked software at the moment. Don't want to go back to canvas. Hardly any infill options in there.

Dandydorf, 2023-01-04:
> This is still being worked on, have received little or no feedback on non-working issues lately, but addressed all of them as they passed.    PS2.5 roll-out was pretty uneventful with no major changes in terms of code generation.   Has some great features but they don’t seem to interact with P2PP..  
> 
> > 
> > 
> > I wish the P2PP team still worked on this. If prusaslicer newer versions were still usable, i would be able to use p2pp right this second.... 2.3 is totally broke

If nothing else comes up, and if i'm the only person with this issue i'll try this again on the shop computer to see if it works there. I can only hope that this is temporary or that theres a new update/version coming soon. Fingers crossed!!

tomvandeneede, 2023-01-04:
About this particular question... it was in at some time but removed because it did not add any value really and with the coming of the 8-color palette it was hard to fit the info in the small window...  the only remainder from this development was that if the settings are still the default (70mm) P2PP will issue a warning on that. ...

tomvandeneede, 2023-01-04:
Don’t worry about it too much,  you’ll get it up and running again.

What would you go back to 2.3 anyway? 2.5 seems to do ok as well (depending on the printer you have maybe, cannot comment on that), just make sure you have an up to date version of P2PP, config wise there should not be much to change after updating either of the softwares.

Dandydorf, 2023-01-04:
> 5

Well man the truth is, i was under the impression that only 2.3 works with P2PP. All documentation (and on youtube) says so as of right now. Maybe you can update that on github but i am all for using 2.5 if i can!

tomvandeneede, 2023-01-04:
2.5 works just fine

Dandydorf, 2023-01-04:
> 2.5 works just fine
> 
> > 
> > 
> > 5
> > 
> > Well man the truth is, i was under the impression that only 2.3 works with P2PP. All documentation (and on youtube) says so as of right now. Maybe you can update that on github but i am all for using 2.5 if i can!

I appreciate you letting me know about that. I just set up all the start gcode and settings for the palette for in the morning, so i'll be well off thanks to your mindfulness. Thanks a bunch!! Should be a lot quicker now lol

tomvandeneede, 2023-01-04:
good to hear will add 2.5 to the list of supported versions o  the site to avoid further confusion