tomvandeneede/p2pp #136: PURGETOPSPEED command is causing pausing in purge tower.

OldSaltyRC, 2022-05-14:
In version 8.00.43. ;P2PP PURGETOPSPEED command is causing the print to randomly pause while purging.  When paused I can press the 'Pause' button on the Palette 3 Pro, then press 'Resume' to get it to start printing again, but eventually this causes the pings to get to far off and the Palette stops.   It's possible that the PURGETOPSPEED command is no longer a valid command, I had to use it back in December to keep the pings from getting out of whack, then I was unable to print up until this last month.  That's when I updated P2PP and the Palette firmware, and developed this pausing issue.  Through trial and error I was able to isolate the pausing issue to the PURGETOPSPEED command.  Once I removed it all together from my start Gcode everything worked fine again. 

Just thought I would mention it here, just in case no one was aware of this issue.

Thanks.

tomvandeneede, 2022-05-15:
Would you mind sharing the gcode file for this print please.  There could maybe be an unwanted side effect in a specific case but would have to look a the gcode (if possible processed and unprocessed gcode so I can rerun P2PP to examine the behaviour

tomvandeneede, 2022-05-16:
> Sure, how shall I send them, I'm only able to upload logs here?
> Uploading log_2022_05_06T04_27_37_599Z.gz… <applewebdata://CD97B746-54BB-4630-B9F7-327DD8665520>

wetramsfer of dropbox usually work best

OldSaltyRC, 2022-05-16:
Sending link to files via FB Messenger.