tomvandeneede/p2pp #138: Unprocessed Gcode

UserWally, 2022-06-03:
I am probably doing something wrong on my end.
I put the  ;P2PP SAVEUNPROCESSED line in the start script but I do not see any stand alone gcode file after processing.
Is there something else I need to do? Using Palette3 Pro

tomvandeneede, 2022-06-03:
output should be an mcfx file, not a gcode file

UserWally, 2022-06-03:
I always see requests for unprocessed files for troubleshooting. What are they?

tomvandeneede, 2022-06-04:
There are tons of settings in Prusa Slicer and P2PP and  some issue occur with specific combinations.   This is by far the easiest way to import those exact same settings for debugging and solutioning.   The saved file will reside in the temporary folder (you can get that from the processing window.
)

closed 2022-06-12 by UserWally