tomvandeneede/p2pp #139: [Bug] Filament Profile Types not getting parsed

rcmaniac25, 2022-06-06:
P2PP Version: 8.00.44
Python Version: 3.9.1
Platform: win32

PrusaSlicer 2.4.2 windows

Palette 3 Pro
Prusa MK3S+

Hi, I found 2 issues:

1. The P3 setup guide says to set `:P2PP FILAMENTOVERRIDE=` to support custom profiles, but this did not work. Looking at the source code, I saw it was `PROFILETYPEOVERRIDE` not `FILAMENTOVERRIDE`
2. Along the way, I realized that parsing only touches the first filament type's start code and ignores all others. This should probably be `i` instead of `0`: https://github.com/tomvandeneede/p2pp/blob/master/p2pp/psconfig.py#L331

I tried to do a local build before making a PR but I got build issues so didn't want to just make a blind PR without testing locally first. But figured I should at least make a issue as you could fix it before I get to my local build working.

rcmaniac25, 2022-06-06:
Ah, I was unaware you didn't have to choose from the drop down. I'll close then

closed 2022-06-06 by rcmaniac25

tomvandeneede, 2022-06-06:
you can even enter a new value in the drop down so you can create e.g.pla2 and use in the splice definitions.

rcmaniac25, 2022-06-06:
Yea, that's what I did.