tomvandeneede/p2pp #140: Feature Request: Palette Connect

cfelicio, 2022-07-14:
Hi There,

Klipper has the P2 integrated mode, but it requires initialization by using the command PALETTE_CONNECT. I tried to add this to the start gcode via the Prusa Slicer, but P2PP will add commands at the very top of the GCODE file, and Klipper errors out with: 

// Palette 2 is not connected, connect first
!! Cannot initialize print, palette 2 is not connected

Is it possible to have PALETTE_CONNECT added to the very beginning of the print, if the parameter ;P2PP KLIPPER_TOOLCHANGE is specified?

Thanks!

XylenC4, 2022-08-25:
It would be a nice feature to add G-Code intentionally on top of the Palette code

I'm doning it using a Batch-Script (I'm running windows)

I'm calling it "AddLinesOnTop.bat" inside the p2pp folder.
```@ECHO OFF
SET MYPATH=%~dp0

echo PALETTE_CONNECT> newFile.txt
copy newFile.txt+%1 combinedFile.txt
move /Y combinedFile.txt %1
del newFile.txt
```

In PrusaSlicer -> Print Settings -> Output Options -> Post-processing add this line AFTER calling PP2P

```
YOUR P2PP FOLDER\p2pp.exe;
YOUR P2PP FOLDER\AddLinesOnTop.bat;
```

I'll had a lot of wasted filaments because of issue on initialization.
- Palette not Initialized
- Palette initialized, but connection is not possible
- Not Pre-Heated
- Issues on Bed-Leveling

As most of the time the initialization fill fail because of a missing connection, no pre-heat or issues on bed levelling I've made a special M-Command to prepare the printer just before the Palette 2 start it's work.

This is my "AddLinesOnTop.bat" inside the p2pp folder.
```@ECHO OFF
SET MYPATH=%~dp0

echo M99 > newFile.txt
copy newFile.txt+%1 combinedFile.txt
move /Y combinedFile.txt %1
del newFile.txt
```

This is my M99 inside my printer.cfg which presents the initialisation of the printer:
This has to be left-out in PrusaSlicer -> Printer Settings -> Custom G-code -> Start G-code

```
[gcode_macro M99]
gcode:
 PALETTE_CONNECT
 PALETTE_CLEAR
 M190 S60 ; wait for bed temperature
 M109 S210 ; wait for nozzle temperature
 G28
 G1 X30 Y30 Z10
 M82 ; absolute extrusion mode
 G21 ;metric values
 G90 ;absolute positioning
 M107 ;start with the fan off
 G29 ;AutoBedLevel
 G1 X100 Y30 Z150
```

tomvandeneede, 2022-08-25:
Hi,

Should not be a big problem to add this, just wondering what the advantage is over adding these lines into the startup code which goes on top of the print anyway?   The commands to add the extra lines will need oto go there as well no? So basically will be more work to add it there unless there is an extra layer there that I am missing.

Happy to further discuss and action

Regards
Tom

cfelicio, 2022-08-25:
I think these are 2 different issues. The one I originally opened is for a specific command / specific setup (klipper), and the line needs to be at the very top, as it ensures the palette is connected prior to sending commands.

XylenC4, 2022-08-26:
Let's say like there is a need for Klipper to add G-Code in front of the Palette code to perform the initialization of the Interface (PALETTE_CONNECT). It would be nice if it can be multi line for doing other stuff as like pre-heating.

Hotwire, 2024-01-08:
> Hi, Should not be a big problem to add this, just wondering what the advantage is over adding these lines into the startup code which goes on top of the print anyway? The commands to add the extra [...]

This isnt so, there is a bunch of code added by P2PP prior to the Start G-Code 👍 

`O21 D0014
O22 D1ddc8a24efda682e
O23 D0001
O24 D0000
O25 D0 D0 D1FF0000RedPETG D10000FFBluePETG
O26 D0005
O27 D0009
O28 D0001
O29 D0000
O30 D2 D4309a9a9
O30 D3 D43e27ab8
O30 D2 D44fa4532
O30 D3 D452e995c
O30 D2 D45457ba2
O32 D11 D0000 D0000 D0000
O1 DPS_2_5_MosaicKeychain_PETG_CR10V3 D00000c58
M0
;Splice Information:
;-------------------
;       Splice Offset =    41.00mm
;       Autoloading Offset =     0.00mm

;0001   Input: 3  Location:     0.00mm   length   137.66mm 
;0002   Input: 4  Location:   137.66mm   length   315.30mm 
;0003   Input: 3  Location:   452.96mm   length  1549.20mm 
;0004   Input: 4  Location:  2002.16mm   length   791.42mm 
;0005   Input: 3  Location:  2793.58mm   length   366.14mm 

;Ping Information:
;-----------------
;Ping 0001 at   331.72mm
;Ping 0002 at   664.30mm
;Ping 0003 at   995.32mm
;Ping 0004 at  1326.33mm
;Ping 0005 at  1657.33mm
;Ping 0006 at  1988.90mm
;Ping 0007 at  2319.94mm
;Ping 0008 at  2651.02mm
;Ping 0009 at  2982.43mm

;------------------------:
; - Process Info/Warnings:
;------------------------:
;Generated with P2PP version 9.01.05
;Processed file:. 
;P2PP Processing time  0.49s
;No warnings

;--------- THIS CODE HAS BEEN PROCESSED BY P2PP v9.01.05 --- 

T0
; generated by PrusaSlicer 2.7.1+win64 on 2024-01-08 at 16:09:46 UTC

;
; thumbnail begin 32x32 1412
; thumbnail end
;

;
; thumbnail begin 400x300 16592
; thumbnail end
;
;

; external perimeters extrusion width = 0.42mm
; perimeters extrusion width = 0.44mm
; infill extrusion width = 0.44mm
; solid infill extrusion width = 0.44mm
; top infill extrusion width = 0.40mm
; first layer extrusion width = 0.42mm

; external perimeters extrusion width = 0.42mm
; perimeters extrusion width = 0.44mm
; infill extrusion width = 0.44mm
; solid infill extrusion width = 0.44mm
; top infill extrusion width = 0.40mm
; first layer extrusion width = 0.42mm

;TYPE:Custom`

All of the above code comes before any of the PrusaSlicer Start G-Code script. 

If PALETTE_CONNECT is added to the start of the file as part of the P2PP processing, perhaps this can help?

Hotwire, 2024-01-08:
As an update, I just tried it.. .

Adding PALETTE_CONNECT as the first line in the file performed perfectly.