tomvandeneede/p2pp #145: Incorrect Splice Settings Applied to PrusaSlicer Parts for Palette 3

veryfairweather, 2022-10-09:
There appears to be a bug that is causing splicing settings to not be applied accurately to objects with multiple parts in PrusaSlicer. 

I am printing using two different types of material, PLA and NGEN. On my Palette 3, I use 2, 2, 4 for PLA. However, the PLA will not stick to NGEN without high heat and compression, so for PLA-NGEN I use 6,6,8. I can easily and consistently confirm the effectiveness of these settings in the splice tuner. Because the PLA-NGEN splice requires high heat and compression, if the splice fails I can tell when that is due to the settings not being applied, because the splicing process is taking less time to heat and cool than it should be.

Here is how I have been able to confirm and replicate this issue. If I add an object (like a cube) to the plater in PrusaSlicer, and then add another cube as a separate object next to it, I can assign an extruder to each object (one using PLA and the other using NGEN) and my settings work. I have no issues whatsoever. 

However, if I add a cube to the plater and then add another cube **_on top of that as a part_** (which is the only way to stack components), the correct settings do not get applied, and my PLA-NGEN splices end up cold splicing and failing. I can tell that less time is spent heating and cooling during the splice process, so I know the incorrect settings are being applied. I also am not receiving any error messages from P2PP. In PrusaSlicer, when you add a part to an existing object, the program creates two parts under one parent object. That parent object has its own extruder setting, and each part has its own extruder setting. I believe the existence of the extruder setting for the parent object is contributing to this issue.

tomvandeneede, 2022-10-10:
Hi,

would you mind sharing the gcode/mcfx files with me so I can take a look?

regards

tomvandeneede, 2022-10-17:
> multimaterial test.zip <https://github.com/tomvandeneede/p2pp/files/9796088/multimaterial.test.zip>
> Uploaded the MCFX file for each example described above, as well as the corresponding 3mf files. Thanks for taking a look!

I see that you have the following filaments setup:

  "filaments": [
    {
      "materialId": 2,
      "filamentId": 1,
      "type": "PLA",
      "name": "DarkSlateBlue",
      "color": "#004080"
    },
    {
      "materialId": 1,
      "filamentId": 2,
      "type": "NGEN",
      "name": "GreenYellow",
      "color": "#00FF40"
    }
  ]

so input 1 contains material type 2 = PLA, input 2 contains material type 1 = NGEN

  "algorithms": [
    {
      "ingoingId": 2,
      "outgoingId": 1,
      "heat": 6,
      "compression": 6,
      "cooling": 8
    }
  ]

further there is 1 also defined going from material type 2 to material type 1 with 6-6-8 parameters

so at first sight all looks ok.   I will check with the Mosaic team is I  have missed anything here