tomvandeneede/p2pp #146: invalid literal for int() with base 10: 'x'

swag012004, 2022-10-21:
I am getting the above error and it is failing to continue with the processing script.

I have a palette 3 pro with an m1 mac.

Thanks

Greg

tomvandeneede, 2022-10-21:
Can you please send me a screenshot of this issue and the g-code that is causing this issue (unprocessed code)
I will try to see what is wrong and fix it

swag012004, 2022-10-21:
sure.  
[image]
[Tigger_Yellow.mcfx.zip](https://github.com/tomvandeneede/p2pp/files/9842051/Tigger_Yellow.mcfx.zip)

Thank you!

tomvandeneede, 2022-10-21:
Tc and Tx commands in your gcode are for MMU2 and unfortunately will not process for Palette
8.0.48 will remove the processing error but you should get them out of the config if you want 100% control over your print.

8.0.48 should be available for download now

swag012004, 2022-10-21:
wow, that was fast!   Thank you.   Let me try it

swag012004, 2022-10-21:
That worked.  Thanks again