tomvandeneede/p2pp #148: Getting error 131 - Timeout Homing Outgoing Drive

DeanDavis, 2022-11-07:
Getting error 131 - Timeout Homing Outgoing Drive

Doing a simple 2 color print.

Printing from Canvas works fine.

[P2PP Test Print_0.2mm_PLA_MK3S_32m.mcfx.zip](https://github.com/tomvandeneede/p2pp/files/9956351/P2PP.Test.Print_0.2mm_PLA_MK3S_32m.mcfx.zip)

Start the job and it loads 1 to the cutter then loads 2 to the cutter. Then it loads 1 to the outgoing drive, cuts it, retracts it and then fails with the 131 error.

tomvandeneede, 2022-11-08:
did you get any warnings of shortnsplices during the processing?

DeanDavis, 2022-11-08:
Yes. P2PP output of this test job is attached.
Since it just was a warning and a test print just thought I'd see how it would print and then try and make adjustments if the print came out bad.

[P2PP Test Print Output.txt](https://github.com/tomvandeneede/p2pp/files/9961692/P2PP.Test.Print.Output.txt)

tomvandeneede, 2022-11-08:
Nah these are warnings that the print won’s probably print correctly 

SHORT FIRST SPLICE (min 130.0mm) Length:116.44 Input 0
SHORT SPLICE (min 90.0mm) Length:84.76 Layer:5 Input:2
SHORT SPLICE (min 90.0mm) Length:84.23 Layer:6 Input:1
SHORT SPLICE (min 90.0mm) Length:84.02 Layer:7 Input:2
SHORT SPLICE (min 90.0mm) Length:83.98 Layer:8 Input:1

The 84/83mm ones likely will pose little issue but the 116 is way short

tomvandeneede, 2024-01-22:
> I have the same issue! P2PP warns the first splice is too short, P3P cuts the first filament first and stops with error 131.

Just add a a few loops around the print to make the first splice longer.   There is no automated way of doing this unless you use the FULLPURGEREDUCTION feature
Any other splices can be fixed by increasing the purge.  a few mm should not be too big of an issue except with the first splice.

cyberhuman, 2024-01-22:
I tried to AUTOADDPURGE and it requires side wipe. Very sad.

> Automatic Splice length increase NOT activated due to incompatible mode
> Automatic Splice length increase works with Full purge reduction and side wipe only

tomvandeneede, 2024-01-22:
I believe fullpurgewreduction also supports this mode.  
IOt is not easy to get rid of extra filament in a pre-generated code file.

cyberhuman, 2024-01-22:
I added some loops to the brim and it helped. Thank you!