tomvandeneede/p2pp #150: PRUSASLICER REFUSES TO OPEN

Dandydorf, 2023-01-04:
ps2.3 refuses to open at all, on a fresh install on windows 10. Any ideas what the issue is? It shows this error and crashes upon startup[image]

Dandydorf, 2023-01-04:
> Did you install PS2.3 or did you install something else?   Looks like a PS issue at first glance, you may have selected marlin2 in a later version of PS and then installed 2.3 again.  2.3 seems to be complaining that if does not know marlin2 so does not start
> P2PP is agnostic to this setting so don’t think it is the source of the problem.

Oh yeah i'm surely saying its a prusaslicer issue but i can't find much info on prusaslicer or how to fix this. Apparently marlin2 is default? I haven't been able to change it from marlin2 anyways even after editing the printer settings. I tried installing PS2.3 and 2.3 'rc3' to see if there was any difference. Also tried reinstalling to no avail.

tomvandeneede, 2023-01-04:
I believe either throw away the setup (this stays between versions), PS will make a full new one, or reinstall a later version and switch to another flavour  that exists in PS2.3, afterwards you can switch back to 2.3