tomvandeneede/p2pp #151: Pause print issue - P2PP + PS2.5.0 + MK3S+

engofthings, 2023-01-20:
Hi,

I have a P3P, using connected mode and Prusa Mk3S+.

I'm trying to pause the print in PrusaSlicer 2.5.0 with the (+) sign on the layer bar (Right click-> Add Pause print M601).

Unfortunately the printhead keeps moving and destroying the print when its time to pause, instead of pausing. I am experiencing very fast extrusion, filament loosing grip, blobs start to appear on the part.

I checked 2 versions (attached):
1.  file P3P.gcode -  P2PP Multicolor with pause @Z1.1 - production file sent by P2PP, print started from P3P - Screwed things up
2.  file MK3S.gcode - Single color with pause @Z1.1 - printing from SD Card, print started from MK3S+ - Prints as expected

It seems like the palette ignores the M601 and continues to send commands.

How can I use P2PP and pause feature? What  am I doing wrong?

[Downloads.zip](https://github.com/tomvandeneede/p2pp/files/10467695/Downloads.zip)

tomvandeneede, 2023-01-20:
Hi,

I still see the M601 for inserting magnets in the code, so it does not appear to be P2PP mistreating this setting.
I suggest asking Mosaic Support if they have specific processing for this command that may throw things off
Unless there is another way to insert a pause I do not see how P2PP can help out here… 

Regards
Tom

engofthings, 2023-01-21:
Hi,

Thanks for answering.

I asked Mosaic, M0 command is supported. ([Link here for P2](https://support.mosaicmfg.com/Guide/How+to+Add+Pauses+for+Filament+Changes/136))
[image]

memilnieAU, 2023-03-25:
Hi, I just tried the pause function and came up with this modification so I can easily use the pause function as usual.

In the Template Custom G-Code add the following code:
G1 Y-3 F1000 ; go outside print area
G91   ; Relative XYZE
G1 Z+50 E-5 F3600  ; Move Z up 50 and retract 5 mm at 60 mm/s (3600/60)
G90   ; Absolute XYZE
M0 ; Pause at Palette

Then it’s passible to right click at the layer that I want to pause at, and pick the Template G-Code

[image]

tomvandeneede, 2023-03-26:
Would this not be more for e slicer feature than a P2PP feature.   Maybe I do not 100% get what you are trying to achieve. … if anything in P2PP it would have to be programmed in the startup recode and it would be very prone to error as there is no way to sync layers with print features.. 

Can you elaborate a little bit further on what exactly you think to achieve?

memilnieAU, 2023-03-26:
The hope was to pause the print after a layer to add some magnets to the print. Normally I will use the Pause function in Prusa Slicer (M601), but as engofthings says it dose not work. Therefore I added a the commands above, to pause the print after a specific layer. 
But I think you have right, Its a Mosaic issue and not P2PP, but I think the code can solve @engofthings' isues =)