tomvandeneede/p2pp #152: IP or Host not found

rnunez24, 2023-01-24:
Using a Palette 3 PRO (accesory mode) klipper cr10 and trying to slice with prusaslicer 2.5.

tomvandeneede, 2023-01-24:
can you ping the palette with this name.

rnunez24, 2023-01-24:
[image]

P2PP Version: 8.00.49
Python Version: 3.9.1
Platform: win32

Reading File C:\Users\ruben\AppData\Local\Temp\.6312.gcode.pp
Analyzing Prusa Slicer Configuration
All purge lenghths 70/70 OR 140. Purge lengths may not have been set correctly.
Linear Ping interval of 350.00mm
SPLICE OFFSET: 30.00mm
Convert to absolute extrusion parameters
Extra filament at end of print 150.00mm
Config: Palette3 Accessory Mode Selected
GCode Analysis ... Pass 1
Tower detected from (169.282, 157.85600000000002) to (238.218, 181.19199999999998)
Automatic Splice length increase activated
Gcode Analysis ... Pass 2
Processing time 0.71s
Generating GCODE file: (temp location, PS will move) C:\Users\ruben\AppData\Local\Temp\.6312.gcode.pp
Generating Palette 3 output files
Generating PALETTE MAFX file: C:\Users\ruben\Downloads\201901MosaicKeychain1_d0.4_58m_0.20mm_210C_PLA_CR10S.mafx

----------------------------------------
Print Summary
----------------------------------------

Number of splices: 5
Number of pings: 8
Total print length 3613.62mm

Inputs/Materials used:
----------------------
Input 1 2636.69mm - PLA [######] Silver SMART_MATERIALS_PLA_CREALITY 
Input 2 976.93mm - PLA [######] Black SMART_MATERIALS_PLA_CREALITY 

Splice Information:
-------------------
Splice Offset = 30.00mm
Autoloading Offset = 0.00mm
0001 Input: 1 Location: 0.00mm length 133.05mm
0002 Input: 2 Location: 133.05mm length 150.65mm
0003 Input: 1 Location: 283.70mm length 2040.95mm
0004 Input: 2 Location: 2324.65mm length 826.28mm
0005 Input: 1 Location: 3150.93mm length 492.69mm

Ping Information:
-----------------
Ping 0001 at 423.54mm
Ping 0002 at 787.81mm
Ping 0003 at 1185.23mm
Ping 0004 at 1582.64mm
Ping 0005 at 1982.36mm
Ping 0006 at 2332.80mm
Ping 0007 at 2775.90mm
Ping 0008 at 3128.43mm

===========================================================================================
Go to https://github.com/tomvandeneede/p2pp/wiki for more information on P2PP Configuration
===========================================================================================
Sending file 201901MosaicKeychain1_d0.4_58m_0.20mm_210C_PLA_CR10S.mcfx to P3 (192.168.0.42)
Uploading 201901MosaicKeychain1_d0.4_58m_0.20mm_210C_PLA_CR10S.mcfx
|██████████████████████████████████████████████████| 31/31Kb [100%]
Upload aborted by user

tomvandeneede, 2023-01-24:
You have to enable this feature in p3 as well did you do that?

rnunez24, 2023-01-24:
> You have to enable this feature in p3 as well did you do that?
> […](#)

> You have to enable this feature in p3 as well did you do that?
> […](#)

Sorry Tom, what feature ?

rnunez24, 2023-01-24:
Sorry Tom, What feature is needed to enable?

tomvandeneede, 2023-01-24:
https://github.com/tomvandeneede/p2pp/wiki/Uploading-files-to-P3￼
Uploading files to P3
github.com

rnunez24, 2023-01-24:
ouhggggg.....local file transfer was OFF :(

now is ok! mcfx in p3p. Thx!!

And what about the .gcode? Where is the .gcode file to upload to klipper?

tomvandeneede, 2023-01-25:
That should be where you exported it

rnunez24, 2023-01-25:
Configuration reference says:

_Output filename format

P2 will use an extension of .mcf.gcode

P3 will use extension of .mcfx_

With this P3 configuracion, any gcode is generated to send to the printer in accessory mode. I solved it adding.gcode after .mcfx

 eg.

{input_filename_base}_{layer_height}mm_{filament_type[0]}_{printer_model}_{print_time}.mcfx.gcode

I don't know if this is considered as an issue?

closed 2023-01-27 by rnunez24