tomvandeneede/p2pp #153: purge tower first layer height

dburden42, 2023-02-28:
When printing with Palette Pro 3, the skirt and brim all print correctly, but the first purge tower layer is printing out almost directly on the print surface. Using PURGETOWERDELTA=10

tomvandeneede, 2023-03-01:
Can you give a little bit more information regarding the versions of slicer/p2pp you are using.

dburden42, 2023-03-01:
Hello, thanks - I'm using PrusaSlicer Version 2.5.0+win64, and p2pp_8.00.49. Here's my P2PP start code in Prusa:

; Railcore 300ZL start code - 20 May 2021

M140 S[first_layer_bed_temperature] ; Raise bed temp to profile requested bed temp
M190 S[first_layer_bed_temperature] ; Wait for bed temp to rise
G28 ; home all axes
G32 ; Autocalibrate bed
G32 ; Autocalibrate bed a second time
G28 Z ; Home Z after bed leveling
;
G1 Z20 F600 ; Move bed down 20
M104 S[first_layer_temperature] T0 ; Raise hot-end temp to profile requested hot-end layer 1 temp
M109 S[first_layer_temperature] T0 ; Wait for hot-end temp to rise to profile requested hot-end layer 1 temp
;
G1 X0.0 Y0.0 Z1.0 F2000.0 ; prepare to prime
G92 E0 ; reset extrusion distance
G1 Z0.2 F600
G1 X60.0 E9.0  F1000.0 ; priming
G1 X125.0 E12.5  F1000.0 ; priming
G1 Z0.5 F600
G92 E0 ; reset extrusion distance

;P2PP PALETTE3_PRO
;P2PP P3_SHOWPRINTERPAGE
;P2PP PRINTERPROFILE=56a564b6077fa7bb84ea1c9097193a28
;P2PP LINEARPINGLENGTH=350
;P2PP SPLICEOFFSET=40
;P2PP MINSTARTSPLICE=150
;P2PP MINSPLICE=90
;P2PP ABSOLUTEEXTRUDER
;P2PP MATERIAL_DEFAULT_0_0_0
;P2PP MATERIAL_PLA_PLA_0_2_0
;P2PP MATERIAL_ABS_ABS_0_0_0
;P2PP MATERIAL_PLA1_PLA2_0_0_0
;P2PP MATERIAL_PLA2_PLA1_0_0_0
;P2PP MATERIAL_PET_PET_0_0_0
;P2PP PURGETOWERDELTA=999
;P2PP WIPEFEEDRATE=3000
;P2PP EXTRAENDFILAMENT=120
;P2PP SAVEUNPROCESSED

; end of start code

tomvandeneede, 2023-03-01:
Ok would you mind sending me the unprocessed gcode for the print so I can simulate the issue

> On 1 Mar 2023, at 15:16, dburden42 ***@***.***> wrote:
> 
> M140 S[first_layer_bed_temperature] ; Raise bed temp to profile requested bed temp
> M190 S[first_layer_bed_temperature] ; Wait for bed temp to rise
> G28 ; home all axes
> G32 ; Autocalibrate bed
> G32 ; Autocalibrate bed a second time
> G28 Z ; Home Z after bed leveling
> ;
> G1 Z20 F600 ; Move bed down 20
> M104 S[first_layer_temperature] T0 ; Raise hot-end temp to profile requested hot-end layer 1 temp
> M109 S[first_layer_temperature] T0 ; Wait for hot-end temp to rise to profile requested hot-end layer 1 temp
> ;
> G1 X0.0 Y0.0 Z1.0 F2000.0 ; prepare to prime
> G92 E0 ; reset extrusion distance
> G1 Z0.2 F600
> G1 X60.0 E9.0 F1000.0 ; priming
> G1 X125.0 E12.5 F1000.0 ; priming
> G1 Z0.5 F600
> G92 E0 ; reset extrusion distance
> 
> ;P2PP PALETTE3_PRO
> ;P2PP P3_SHOWPRINTERPAGE
> ;P2PP PRINTERPROFILE=56a564b6077fa7bb84ea1c9097193a28
> ;P2PP LINEARPINGLENGTH=350
> ;P2PP SPLICEOFFSET=40
> ;P2PP MINSTARTSPLICE=150
> ;P2PP MINSPLICE=90
> ;P2PP ABSOLUTEEXTRUDER
> ;P2PP MATERIAL_DEFAULT_0_0_0
> ;P2PP MATERIAL_PLA_PLA_0_2_0
> ;P2PP MATERIAL_ABS_ABS_0_0_0
> ;P2PP MATERIAL_PLA1_PLA2_0_0_0
> ;P2PP MATERIAL_PLA2_PLA1_0_0_0
> ;P2PP MATERIAL_PET_PET_0_0_0
> ;P2PP PURGETOWERDELTA=999
>

dburden42, 2023-03-02:
Hello again,

I just realized you asked for the unprocessed gcode, but for some reason my modified start gcode doesn't produce an unprocessed file, even though I've included ;P2PP SAVEUNPROCESSED