tomvandeneede/p2pp #154: P2PP sending moves out of range, even with new print file.

Dandydorf, 2023-03-03:
Does this sound like a prusaslicer issue, or something to do with the printer itself? I have also had a myriad of random issues such as

The palette put out a goddamn unlimited amount of filament on the print start (still trying to figure this out) and on the same token, the palette unrelatedly undershoots the filament load amounts by more than 5%. 

The one print i did get going, ended up failing right before the first splice and couldn't get past an 'unhandled exception' which is apparently uncommon. Should an update  of PS or p2 fix ANY of this? It feels so fucking hopeless loading and unloading 3 feet of filament at a time to waste it.

tomvandeneede, 2023-03-03:
what version of ps and p2pp are you using. can you send me a screenshot of the error as well as the unprocessed gcode so I can simulate the issue

Dandydorf, 2023-03-03:
I have gcode but i believe i don't have any unprocessed. I have seen that there is a new version of P2pp out and it may fix a few of these issues

tomvandeneede, 2023-03-03:
Please try the latest version first then.

Regards

Dandydorf, 2023-03-11:
Just tried resending the 3d file, will try upgrading the version next