tomvandeneede/p2pp #155: Error in P2PP 9.00.000

Darkside2016, 2023-03-05:
We're sorry but an unexpected error occurred while processing your file
Please sumbit an issue report on https://github.com/tomvandeneede/p2pp

Error: '

Traceback Info:
File "p2pp.py", line 115, in 
File "C:\Users\tomvandeneede\Documents\p2pp\_build_update_\p2pp\p2pp\mcf.py", line 1043, in p2pp_process_file 
File "C:\Users\tomvandeneede\Documents\p2pp\_build_update_\p2pp\p2pp\mcf.py", line 838, in parse_gcode_second_pass

tomvandeneede, 2023-03-06:
Can you please send me the gcode file for this print

Darkside2016, 2023-03-06:
[BEAR 3MF 02_2d4h15m_0,20mm_205C_PLA_CR10S.zip](https://github.com/tomvandeneede/p2pp/files/10896609/BEAR.3MF.02_2d4h15m_0.20mm_205C_PLA_CR10S.zip)
Hi,
attached is the file that Prusa created.

This error that I have indicated only occurs since version 9.x.
I do not have this error in version 8.x of P2PP.

P2PP also cannot complete the process because of the error.

tomvandeneede, 2023-03-06:
Can you please also include the unprocessed code

Darkside2016, 2023-03-06:
[BEAR 3MF 02 Color.zip](https://github.com/tomvandeneede/p2pp/files/10896786/BEAR.3MF.02.Color.zip)

This is the finished file with the colours to be used.

tomvandeneede, 2023-03-06:
really need the gcode file, 3mf is of no help as it depends on versions etc

Darkside2016, 2023-03-06:
P2PP cannot finish the file because of the error.
The first file I sent is what Prusa saves first and then P2PP opens.
After a little while the error appears and P2PP hangs at 78% and does not continue.

Darkside2016, 2023-03-06:
[BEAR 3MF 02_0.16mm_PLA_CR10V2_2d6h42m.zip](https://github.com/tomvandeneede/p2pp/files/10897095/BEAR.3MF.02_0.16mm_PLA_CR10V2_2d6h42m.zip)

I have now taken out everything that concerns P2PP.

However, I would also be happy to send you my custom Gcode.

;P2PP PALETTE3_PRO
;P2PP P3 PROCESSPREHEAT

M201 X500.00 Y500.00 Z100.00 E5000.00 ;Setup machine max acceleration
M203 X500.00 Y500.00 Z10.00 E50.00 ;Setup machine max feedrate
M204 P500.00 R500.00 T500.00 ;Setup Print/Retract/Travel acceleration
M205 X8.00 Y8.00 Z0.40 E5.00 ;Setup Jerk
M220 S100 ;Reset Feedrate
M221 S100 ;Reset FlowrateG90 ; use absolute coordinates

M83 ; extruder relative mode
M104 ; set temporary nozzle temp to prevent oozing during homing
M140 S{first_layer_bed_temperature[0]} ; set final bed temp
G4 S30 ; allow partial nozzle warmup
G28 ; home all axis

M420 S1 ;Load automatic mesh
G29 L0

M104 S{first_layer_temperature[0]} ; set final nozzle temp
M190 S{first_layer_bed_temperature[0]} ; wait for bed temp to stabilize
M109 S{first_layer_temperature[0]} ; wait for nozzle temp to stabilize

G92 E0 ;Reset Extruder
G1 Z2.0 F3000 ;Move Z Axis up
G1 X10.1 Y20 Z0.28 F5000.0 ;Move to start position
G1 X10.1 Y200.0 Z0.28 F1500.0 E15 ;Draw the first line
G1 X10.4 Y200.0 Z0.28 F5000.0 ;Move to side a little
G1 X10.4 Y20 Z0.28 F1500.0 E30 ;Draw the second line
G92 E0 ;Reset Extruder
G1 Z2.0 F3000 ;Move Z Axis up

;Put printing message on LCD screen
M117 Printing...

;P2PP PRINTERPROFILE=77c19c195bdaccd9098d24740f1c5167

;P2PP SPLICEOFFSET=30
;P2PP MINSTARTSPLICE=130
;P2PP MINSPLICE=90

;P2PP EXTRAENDFILAMENT=150

;P2PP LINEARPINGLENGTH=350

;P2PP MATERIAL_DEFAULT_2_0_1
;P2PP MATERIAL_PLA_PLA_2_0_1
;P2PP MATERIAL_PLA_PLA2_-1_0_0
;P2PP MATERIAL_PLA2_PLA_0_0_0

;P2PP ABSOLUTEEXTRUDER

;P2PP TEMPERATURECONTROL

;P2PP SIDEWIPELOC=X319
;P2PP SIDEWIPEMINY=45
;P2PP SIDEWIPEMAXY=45
;P2PP WIPEFEEDRATE=200
;P2PP SIDEWIPECORRECTION=1.0
;P2PP SIDEWIPEZHOP=3.0

;P2PP CHECKVERSION

;P2PP CONSOLEWAIT

;P2PP AUTOADDPURGE

;P2PP P3_PROCESSPREHEAT

;P2PP P3_UPLOADFILE
;P2PP P3_HOSTNAME=FKCP-P3P

;P2PP P3_SHOWPRINTERPAGE

tomvandeneede, 2023-03-06:
compiling 9.0.1 as fix for testing

bavery79, 2023-07-31:
Would just like to add to this, since I'm still seeing this same error message with 9.01.04.  I suspect it has something to do with a change on the PrusaSlicer side of things, since I'm only seeing the error in the 2.6 release version of PrusaSlicer.  PrusaSlicer 2.6.0-rc2 does not trigger this error.

tomvandeneede, 2023-08-01:
could you try resending the unprocessed fime as I was not able to download is due to access restrictions on the google drive

bavery79, 2023-08-06:
Hey, sorry missed this earlier. Shortly after submitting the original
GitHub ticket I edited it to replace the original drive link with a file
small enough to just attach via GitHub. Were you able to see this? If
you're just working through the email notifications from GitHub maybe you
never even saw the revised edit.