tomvandeneede/p2pp #157: purge tower first layer height 0,7mm

JamesBond0815, 2023-03-25:
When printing with P2PP Purgetowerdelta, the first layer is printed at 0.7mm in air.
PrusaSLicer 2.6.0-alpha5 / P2PP 9.00.01 / Caribou MK3S 220 / Palette 3Pro

[201901MosaicKeychain1_0.2mm_GelbPLAREDLINE_Caribou_MK3S_220_42m.mcfx.zip](https://github.com/tomvandeneede/p2pp/files/11069301/201901MosaicKeychain1_0.2mm_GelbPLAREDLINE_Caribou_MK3S_220_42m.mcfx.zip)
[8992.zip](https://github.com/to
[image]
mvandeneede/p2pp/files/11069302/8992.zip)
[Caribou 220 Gelb.zip](https://github.com/tomvandeneede/p2pp/files/11069316/Caribou.220.Gelb.zip)

tomvandeneede, 2023-03-26:
which version of p2pp are you using?

tomvandeneede, 2023-03-26:
Can you send me the unprocessed gcode please so I can try to figure out what goes wrong.

JamesBond0815, 2023-03-26:
[8992.zip](https://github.com/tomvandeneede/p2pp/files/11071230/8992.zip)
P2PP Version: 9.00.01
PrusaSlicer: 2.6.0 Apha 5
Caribou MKS3S 220

JamesBond0815, 2023-03-26:
P2PP Version: 9.00.01
PrusaSlicer: 2.6.0 Apha 5
Caribou MKS3S 220

tomvandeneede, 2024-02-02:
closed after no feedback received ... assuming issue solved

closed 2024-02-02 by tomvandeneede