tomvandeneede/p2pp #158: Prusaslicer 2.6.0 alpha4 exports a gcode file instead of mcfx

dburden42, 2023-04-04:
I'm trying to implement P2PP with the Prusaslicer 2.6.0 alpha4, but exporting gcode doesn't generate a .mcfx file (even though it has the .mcfx extension). 

Here's two files that I exported from the same project, one from Prusaslicer 2.5.0, and one from Prusaslicer 2.6.0 alpha4
[image]

tomvandeneede, 2023-04-05:
Deos the p2pp screen popup in 2.6.0?

dburden42, 2023-04-05:
Yes, the screen pops up and seems to have the same content regardless of which version of Prusaslicer I am using.

Killerskyman, 2023-06-26:
looks like you don't have view file extensions enabled. 
If you look at the 'type' column, file explorer says that top one is gcode. Also the file icons (left side) are different even though it looks like those files have the same extension. File explorer bases both of those off of the file extension that is there. I think that bottom *.mcfx file is actually named as *.mcfx.mcfx and the top is named *.mcfx.gcode. prusa 2.6.0 seems to force the use of .gcode at the end of the export name.