tomvandeneede/p2pp #160: The value 'extruder1' is not valid for EXTRUDER error using klipper

gbrmatt, 2023-04-25:
I get this error and the print stops, but the printer stays "active" and hot.

The value 'extruder1' is not valid for EXTRUDER

[test mos.gcode.txt](https://github.com/tomvandeneede/p2pp/files/11325354/test.mos.gcode.txt)

I have a feeling this is a klipper related error as it used to work fine and I've not changed anything p2pp related but I'm no expert so I thought I'd get your opinion. 

its just printing the simple mosaic test key chain and this happens just before the layer change, not the colour change, that goes fine.

tomvandeneede, 2023-04-25:
can you please leave all extruder ids to the default settings in klipper.  Klipper named extruder supper has given me headaches before 🫤

gbrmatt, 2023-04-26:
thanks for the reply, where is the extruder ID found? i cant see it in any config file.

tomvandeneede, 2023-04-26:
do you have

;P2PP KLIPPER_TOOLCHANGE

in your startup code?

gbrmatt, 2023-04-26:
Yes, I do.

tomvandeneede, 2023-04-26:
does not look like this is an issue with P2PP