tomvandeneede/p2pp #161: Color Issue

anthony7501, 2023-04-30:
I am using a Palatte 3 with a PRUSA MK3S+ printer. When I use Canvas I don't have any color issues. When I use P2PP I have this issue. No one in the FB P2PP community has any idea how to solve this and I am at a loss. Does anyone Have any ideas?

[image]

[image]

Can someone PLEASE help me. 

Purging Volumes is set to 300.
;P2PP PALETTE3
;P2PP P3_PROCESSPREHEAT
;P2PP P3_UPLOADFILE
;P2PP P3_HOSTNAME=03m5-p3.local
;P2PP PRINTERPROFILE=10b636a301641a5438e9f3f8a842c3d2
;P2PP LINEARPINGLENGTH=350
;P2PP SPLICEOFFSET=30
;P2PP MINSTARTSPLICE=130
;P2PP MINSPLICE=90
;P2PP ABSOLUTEEXTRUDER
;P2PP MATERIAL_PLA_PLA_0_1_1
;P2PP EXTRAENDFILAMENT=100
;P2PP CHECKVERSION

anthony7501, 2023-04-30:
Please let me know if there is any other information you need from me.

tomvandeneede, 2023-04-30:
Can uou send enthe gcode file (unprocessed by p2pp) .  This is very weird indeede

anthony7501, 2023-04-30:
[UN processed color test 4 inch Final Amy Beach_0.3mm_PLA_MK3S_22h39m.mcfx.zip](https://github.com/tomvandeneede/p2pp/files/11361187/UN.processed.color.test.4.inch.Final.Amy.Beach_0.3mm_PLA_MK3S_22h39m.mcfx.zip)
[image]
This pic is using Canvas.

anthony7501, 2023-04-30:
Here you go!

anthony7501, 2023-04-30:
Also when I use Canvas my pings are 99-100  when I use P2PP my pings are 105 consistently.

tomvandeneede, 2023-04-30:
but youmemtion brown hair while the print has a single color head and a coloured jacket

anthony7501, 2023-04-30:
I mean brown dress and is in PRUSA slicer screenshot.

anthony7501, 2023-04-30:
[image]
[image]

anthony7501, 2023-04-30:
Another example of the color not in the right place.

tomvandeneede, 2023-05-01:
Which is weird as the colour changes come in way after they should. So overextrusion should probably bring things forward, not backward

anthony7501, 2023-05-01:
[image]
now this is from Canvas.
[image]

anthony7501, 2023-05-01:
Canvas is working for me, but rather use Prusa Slicer. But it's all weird.

tomvandeneede, 2023-05-01:
The 105 vs 100 could be because of slicer extruder settings.   I wonder though why in your print it does not show any color in the blazer/jacket but startes alternating in the hair… I believe it may be due to the slices being very short (shorter than in Canvas) 

This is what I see in the preview of PrusaSlicer:
￼

This is not the model you are trying to print right?   This model corresponds to the one in your failed print version though but it is weird that you are printing hair /face while it is not in the model and are printing the jacket in skin color while that has a different color in the model…

anthony7501, 2023-05-01:
it's a different model, just showing that anything I print in Canvas is fine. Can I make the slices longer in Prusa slicer, if yes what setting would let me do that? Is that the P2PP MATERIAL_PLA_PLA_0_1_1

tomvandeneede, 2023-05-01:
Nope it is the purge setting.   But in the file you sent me there were no more changes after the jacket… somehow puzzled how it is printing hair wrongly and no jacket at all...

anthony7501, 2023-05-01:
Do you have a test file from a PRUSA I can try? I will test whatever I need to.

anthony7501, 2023-05-01:
Well I do have Canvas working. But let me know if you want me to try anything. Again, I would love to still try PRUSA Slicer.

Kelvinator78, 2023-07-08:
Try adjusting the flow. I haven't tried it before but just thought of it